jacopotediosi/OctoPrint-Telegram #5: Faulty status after connect

derSkimmy, 2016-02-24:
If connecting to an active octoprint-session, and going up 10mm on Z without homing, the bot automatically messages:

`Printing at Z=-10.0.
Bed 26.0/0.0, Extruder 21.9/0.0.
00:00:00, 0% done, 00:00:00 remaining.`

While all messages are customized to german language, this message is probably somewhere hardcoded and a little bit missleading.

fabianonline, 2016-02-25:
It was definitely a bug that you got a message. That should only happen on ZChange events *during* printing.
The message is the default ZChange message ("Message at new height"). Did you customize these as well?

derSkimmy, 2016-02-25:
Yeah, it is all customized to the german counterparts.

fabianonline, 2016-02-25:
Okay, after updating the plugin you'll have to set the ZChange message again, but then it will be used. ;-)

commit 7ef3ae7c (2016-02-25): Fixed a typo in the jinja-template, which prevented you from setting an individual ZCHange message. Fixes #5.
closed 2016-02-25 by fabianonline