jacopotediosi/OctoPrint-Telegram #8: A message after the first layer was printed

GeorgHo, 2016-03-01:
A message after the first layer was printed, to see if it holds.

Many thanks,
Georg

fabianonline, 2016-03-02:
That's a nice idea, although I'm not sure how to detect this... A simple "height is bigger than zero" wouldn't do it, since my start GCodes are "Home to 0, Move to z=5, heat, Move to z=0, wipe, Move to z=0.3 to start the first layer". I think that would have to be something like "the first layer that took more than 3 (?) minutes to complete" or something... I'll look at that.

fabianonline, 2016-03-10:
Another possible way would be to add another configuration option to set fixed heights to send notifications at. You could then say "Always after the first 0.3mm, then after each 5mm".

GeorgHo, 2016-03-11:
Hi Fabian,
this would be a great solution and that works for everyone!

shaiss, 2016-06-03:
could you send a message based on a comment in the gcode?  I know cura writes LAYER=2 to the gcode output

derpicknicker1, 2016-09-24:
@shaiss That is not possible since octoprint does not pass out gcode comments while parsing the gcode... as far as i know

derpicknicker1, 2016-10-01:
Even Gina has no plan for solving it: https://github.com/foosel/OctoPrint/issues/320

wormiedk, 2018-02-09:
Would it be possible to add a notification when the printer STARTS printing the first layer e.g. when heating is complete?

marcboivin, 2018-02-19:
The START notification is definitely within reach.  

The first layer thing is pretty hard, but I might try something because it would be such a great feature. 

Not sure yet I'm comfortable enough to mess with the core code base, so I might try it with a plugin first, see where it goes.

marcboivin, 2018-02-20:
Did a bit of digging. Turns out Cura has comments that identify layers. 

Also with the use of the preprocessor hook, we can take note of all the layers and other comments

Challenge here is to know how we can use those comments while printing to trigger events. 

I'll dig a little deeper see if there's a good way to bring it all together

manos, 2020-04-16:
This might help.. how about just one notification after N mins of printing?
The main thing I want to see, is if the first layer is sticking properly. I don't have a need to get images/notifications every N minutes :)