jacopotediosi/OctoPrint-Telegram #11: No reply from /status when not printing

wizard04wsu, 2016-03-05:
If nothing's printing and I send the /status command, there is no response. It looks like the `TelegramSendNotPrintingStatus` event isn't being handled in [\_\_init\_\_.py](https://github.com/fabianonline/OctoPrint-Telegram/blob/master/octoprint_telegram/__init__.py#L346) (around line 346).

fabianonline, 2016-03-05:
Could it be possible that Octoprint isn't connected to your printer at that
time? That was a condition that I didn't anticipate.
That bus is already fixed in the current development version but not pushed
to be an "official" new version.

wizard04wsu, 2016-03-06:
I tried it both connected and disconnected; neither state gives a response.

fabianonline, 2016-03-10:
Could you please update to 1.1.0 (just released it), enable debug logging (in the plugin's setting, way down at the bottom), try again and then send me the octoprint.log? Thanks.

fabianonline, 2016-03-23:
@wizard04wsu ping?

wizard04wsu, 2016-03-23:
Sorry--thanks for the ping!

I'm not getting any notifications and the only reply from `/status` is if the printer isn't connected.

[octoprint.log.txt](https://github.com/fabianonline/OctoPrint-Telegram/files/187041/octoprint.log.txt)
[chat log.txt](https://github.com/fabianonline/OctoPrint-Telegram/files/187040/chat.log.txt)

I'm still using OctoPrint 1.2.9 (1.2.10 was giving me other troubles so I reverted).

fabianonline, 2016-03-24:
I've just pushed version 1.1.2 which fixes this bug. Thanks for the logs, they were really useful.

commit 747e2cd9 (2016-03-24): Prepared the code to accept the fact that no bed temperatures may be given with _printer. This fixes #11.
closed 2016-03-24 by fabianonline