jacopotediosi/OctoPrint-Telegram #12: Could not save settings for plugin Telegram Notifications (1.0.0)

Fetto, 2016-03-09:
Running Version: 1.3.0.dev895+gb1711f3 (devel branch)

After I've saved the token, I can't save anything else. If i delete the token, I can save the other settings for Telegram. Maybe because my token contains a ":"?

2016-03-09 21:38:28,053 - octoprint.server.api.settings - ERROR - Could not save settings for plugin Telegram Notifications (1.0.0)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev895_gb1711f3-py2.7.egg/octoprint/server/api/settings.py", line 300, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 225, in on_settings_save
    data['token'] = data['token'].strip()
KeyError: 'token'

fabianonline, 2016-03-10:
The ':' in the token has to be there and should be fine.

Could you please update to version 1.1.0 (I released it just now) and - if the problem still exists - enable debug logging in the settings (way down at the bottom), try saving the token and then send me your octoprint.log? That would help me lot.

Fetto, 2016-03-10:
2016-03-10 19:54:40,488 - octoprint.plugins.telegram - ERROR - Not saving token because it doesn't seem to have the right format.
2016-03-10 19:54:40,493 - octoprint.server.api.settings - ERROR - Could not save settings for plugin Telegram Notifications (1.1.0)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev895_gb1711f3-py2.7.egg/octoprint/server/api/settings.py", line 300, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 246, in on_settings_save
    if not data['tracking_activated']:
KeyError: 'tracking_activated'

I have to remove it manually from config.yaml.
Then, when I try to enable debugging and save:
2016-03-10 19:59:37,976 - octoprint.server.api.settings - ERROR - Could not save settings for plugin Telegram Notifications (1.1.0)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev895_gb1711f3-py2.7.egg/octoprint/server/api/settings.py", line 300, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 240, in on_settings_save
    data['token'] = data['token'].strip()
KeyError: 'token'

I tried to enable more logging, but I'm not really sure how to do so.
I tested with:
loggers:
  octoprint.plugins.telegram: DEBUG
but no luck.

Best regards,
Patrik

commit f60743d5 (2016-03-11): Defensive handling of the keys in on_settings_save added. This has the side effect of being compatible to the current devel branch of Octoprint. And it fixes #12.
closed 2016-03-11 by fabianonline