jacopotediosi/OctoPrint-Telegram #14: No answer by the bot

mbreuch, 2016-04-07:
I do get the messages when the Octopi is started or while printing like i set it up, but the bot does not respond to my requests. Neither the abort request nor the help one. 
Did i do anything wrong?

fabianonline, 2016-04-09:
Quite possibly not. Only way to have the bot not talking to you would be having no or a wrong chat id. But in that case you wouldn't get any messages at all.

So something's going wrong in the code. I'd guess the ListenerThread has died... Can you send me your octoprint.log? The error should be somewhere in there.
Also, could you have a look at your telegram history and check what the last successful and the first non-successful command to the bot were and also send me those two? Thanks.

Anyway, after restarting Octoprint, the plugin should be working again. (Until that bug happens again...)

fabianonline, 2016-04-21:
@mbreuch Ping?

mbreuch, 2016-04-25:
The bot did not respond from second 0. I set it up and it sends me all the
notifications, but just does not respond.
Here is a log (I hope it is a right one since am new to all of this).
Greetings

PS: sorry for not responding earlier... been quite busy

fabianonline, 2016-04-26:
Thanks... but unfortunately you either fogot the attachment or github removes attachments from issue comments sent by mail... Could you please upload the log at github.com?

mbreuch, 2016-04-27:
Sure, Here we go, Hope it did work this time
[Octoprintlog_mbreuch.zip](https://github.com/fabianonline/OctoPrint-Telegram/files/238099/Octoprintlog_mbreuch.zip)

mbreuch, 2016-05-01:
Did a complete reinstall since i had some more time this weekend. Now it works like a charm

WHAAAT, 2016-05-09:
i have the same issue... the plugin worked for 1-2 weeks and suddenly stopped answering my requests. however updates are still send

fabianonline, 2016-05-10:
@WHAAAT As soon as I get my new Raspberry (I fear my printer killed my old one...) I'll have a look at the code. The log @mbreuch sent me shows the TelegramListenerThread dieing because it apparently had no internet connection at the time the plugin started. Could you please add your octoprint.log as well so I can have a look into that as well?

WHAAAT, 2016-05-10:
@fabianonline 
great thank you. The plugin itself gives me this exception

`The current connection status is: Got an exception while initially trying to connect to telegram: ('Connection aborted.', gaierror(-2, 'Name or service not known'))`

The log is attached
[octoprint.log.zip](https://github.com/fabianonline/OctoPrint-Telegram/files/257122/octoprint.log.zip)

WHAAAT, 2016-05-11:
i've added the option to log the plugin too. here is my log since restart.
BTW: reinstallation of the plugin doesnt do anything

```

2016-05-11 15:45:04,703 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 1.1
2016-05-11 15:45:04,830 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-05-11 15:45:04,912 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.178.42
2016-05-11 15:45:59,562 - octoprint.plugin.core - WARNING - Trying to mark an unknown plugin telegram
2016-05-11 15:45:59,563 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins and installed plugin packages...
2016-05-11 15:45:59,707 - octoprint.plugin.core - INFO - Found 10 plugin(s) providing 9 mixin implementations, 5 hook handlers
2016-05-11 15:45:59,710 - octoprint.plugin.core - INFO - 10 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/announcements
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/cura
|  Custom Control Editor (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_customControl
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/softwareupdate
|  STL Viewer (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stlviewer
| !Telegram Notifications (1.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/virtual_printer
|  Yamlpatcher (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_yamlpatcher
2016-05-11 15:46:09,445 - octoprint.server.api - INFO - Performing command: sudo shutdown -r now
2016-05-11 15:46:10,906 - octoprint.server - INFO - Shutting down...
2016-05-11 15:46:10,962 - octoprint.server - INFO - Goodbye!
2016-05-11 15:46:28,797 - octoprint.server - INFO - Starting OctoPrint 1.2.11 (master branch)
2016-05-11 15:46:28,877 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins and installed plugin packages...
2016-05-11 15:46:30,122 - octoprint.plugin.core - INFO - Found 10 plugin(s) providing 10 mixin implementations, 6 hook handlers
2016-05-11 15:46:30,134 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2016-05-11 15:46:30,848 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2016-05-11 15:46:31,008 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2016-05-11 15:46:33,284 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 1.1
2016-05-11 15:46:33,289 - octoprint.plugin.core - INFO - Initialized 10 plugin implementation(s)
2016-05-11 15:46:33,996 - octoprint.plugin.core - INFO - 10 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/announcements
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/cura
|  Custom Control Editor (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_customControl
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/softwareupdate
|  STL Viewer (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stlviewer
|  Telegram Notifications (1.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/virtual_printer
|  Yamlpatcher (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_yamlpatcher
2016-05-11 15:46:33,999 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2016-05-11 15:46:34,027 - octoprint.filemanager - INFO - Added 1 items from storage type "local" to analysis queue
2016-05-11 15:46:34,030 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2016-05-11 15:46:34,060 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2016-05-11 15:46:35,037 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2016-05-11 15:46:35,058 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
[... cut ...]
2016-05-11 15:47:09,762 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.178.42
2016-05-11 15:47:16,337 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAAAAAAaaaA1aAAAAaaaaA1aAaa1aAaaaa
2016-05-11 15:47:16,767 - octoprint.plugins.telegram - DEBUG - getMe returned: {u'ok': True, u'result': {u'username': u'TjPrintrBot', u'first_name': u'TjPrintr', u'id': 197798579}}
2016-05-11 15:47:16,769 - octoprint.plugins.telegram - DEBUG - getMe status code: 200
```

WHAAAT, 2016-05-18:
just for information: 1.3.0 did not fix this issue for me. 
(However the new changes are neat)

derpicknicker1, 2016-05-18:
Do you have a log again. So we can see if anything changed.

It may caused by the bot? Have you tried to set up a new one at @BotFather and tested with it. In the past i read a few times about bots which stopped working. My bot for example does not receive special info messages from groups anymore and i can't get the problem solved. So i created a new bot. Changed my settings to new token and all worked fine again. So maybe the problem is not in the code but on the bot API.

But thats no excuse for us. We keep on trying to solve the problem. :)

derpicknicker1, 2016-05-19:
Another idea: 
SSH your pi and sudo raspi-config. 
4. point in list is: "Choose whether to wait for network during boot". 
Maybe this will help.

[Information on raspi-config](https://www.raspberrypi.org/documentation/configuration/raspi-config.md)

derpicknicker1, 2016-09-24:
Is your Problem solved now or are there still problems?

mbreuch, 2016-09-26:
I just do not turn off the raspberry... since then it works all fine

derpicknicker1, 2016-10-01:
Seems like this is not a bug anymore. So i close it.

closed 2016-10-01 by derpicknicker1