jacopotediosi/OctoPrint-Telegram #15: [feature] cam-picture

MicrowaveX, 2016-04-08:
Hi,

i'm quite excited about your plugin! Great work!
I experimented with your plugin-code and added a feature (/campic) which just sends a picture of the printer's cam regardless of the current printing state.

Not sure if you wanna implement it:

`                                                        elif command==gettext("/campic"):
                                                                self.main.track_action("command/campic")
                                                                self.main._send_msg("Ok, here we go you little stalker...", with_image=True)
`

fabianonline, 2016-04-09:
Hi.
Thanks a lot for your support. But did you have a look at the `/status` command? That does essentially the same, just with a little bit more status information... ;-)

MicrowaveX, 2016-04-09:
Yes, but it doesn't send you a pic when you're not connected to the printer ^^

fabianonline, 2016-04-15:
Okay, I don't know why I did it that way, but it seems kinda silly to me now. I'll just change that so you won't need to have another command.

MicrowaveX, 2016-04-17:
Ah ok cool! Great :)

closed 2016-04-17 by MicrowaveX

fabianonline, 2016-04-21:
This feature is included in version 1.2.0 which I've just pushed to github.