jacopotediosi/OctoPrint-Telegram #16: Support for multiple chat-id

alexbarcelo, 2016-04-11:
I will be happy to implement this myself and do a pull request. (I was planning to do this same bot before it existed).

My idea is to allow the user to put a comma-separated list of `chat-id` like:

    2400321, 930323

And allow commands from any of the specified chats.

Do you think that this would be a good approach? It needs minimal changes to the UI. If it makes sense, I will proceed to work on it.

fabianonline, 2016-04-15:
Thanks for your offer.
I already did some of the work for that in the feature/chats branch. I not only want to be able to add more than one chat-id, but also have the ability to send messages to groups. And I want to get away from "look for the id in this list and copy the id into this textbox" to something more GUI-y.

I've already changed the code that searches for unknown chat_ids to also include groups. And I changed the settings template to show a table, one row for each chat and checkboxes for "send notifications to this chat" and "listen to commands from this chat" (this one's disabled for chats).

What's missing is the "save the settings" part and the "send messages to all configured chat_ids" part.

alexbarcelo, 2016-04-15:
Oh! Should have checked the branches before opening my mouth :-|

Glad to see that you are working on it. I think that those ideas would be very nice additions, and I can't wait to use all of them, hehe.

derpicknicker1, 2016-04-26:
See this pull request: https://github.com/fabianonline/OctoPrint-Telegram/pull/19

derpicknicker1, 2016-05-17:
Released in 1.3.0
https://github.com/fabianonline/OctoPrint-Telegram/releases/tag/1.3.0

closed 2016-05-17 by derpicknicker1

Abduhalilbek, 2019-10-14:
rtytrytryty