jacopotediosi/OctoPrint-Telegram #17: [feature request] Send bot file and start a print

mwhuss, 2016-04-15:
Thanks for the great bot, I use it to monitor prints while at work. A feature I'd think would be neat is the ability to send the bot a .gcode file and start a print. If I could do that then I'd be able to close the ports to the web interface externally and do everything through the bot.

fabianonline, 2016-04-21:
This feature is included in version 1.2.0, which I've just pushed to github.

closed 2016-04-21 by fabianonline

mwhuss, 2016-04-21:
Thanks!

[image]

mwhuss, 2016-04-22:
I see the files get uploaded to`uploads/telegram_uploads`, but then they don't show up in /list. Shouldn't they just go straight to `uploads`, or maybe the `watched` folder?