jacopotediosi/OctoPrint-Telegram #20: Check integration into PluginManager

fabianonline, 2016-05-10:
"Installed: 1.1.2, available 1.2.1" => "Everything is up to date."

derpicknicker1, 2016-05-13:
This is the json-response of version check. Why is there an 'v' in information.remote.value? I checked teh respones from all other plugins. They all only contain digits and points in the value field. Also checked the release pages of different plugins. They all only have digit+points version numbers. A short look at code of softwareUpdater seems like it can only handle version numbers vith digits and points (not sure).

So did you, fabian, insert the v? ist it possible to remove it? Did update ever work?
Why did you begin with the 'v' prefix at release 0.15.0?

```
information:
    local: 
        name: "1.1.2",
        value: "1.1.2"
    remote:
        name: "1.2.1"
        release_notes: "https://github.com/fabianonline/OctoPrint-Telegram/releases/tag/v1.2.1"
        value: "v1.2.1"
```

Also there seems to bee a failure in the update data settings for this plugin.

URL is: `pip="https://github.com/fabianonline/OctoPrint-Telegram/archive/{target}.zip"`

but in other plugins: `pip="https://github.com/fabianonline/OctoPrint-Telegram/archive/{target_version}.zip"`

This means:
`{target_version} `instead of `{target}`

derpicknicker1, 2016-05-17:
Solved for newer versions. See here: #23

closed 2016-05-17 by derpicknicker1