jacopotediosi/OctoPrint-Telegram #21: Run system commands

fabianonline, 2016-05-13:
Any way to send system commands?  IE w the systems command editor I now have a power on/off menu item.  This triggers a gpio command to turn on a relay and power the printer.

derpicknicker1, 2016-05-13:
Will custom gCode-commands solve the problem? Like M80 / M81?
Directly sending system commands is a security issue, i think. 
Maybe the known commands from system command editor can be dinamically included to the telegram plugin. Should be possible.

derpicknicker1, 2016-05-14:
Comming soon:
https://github.com/derpicknicker1/OctoPrint-Telegram/commit/eb4808a7efb8fe9ce0eebdd9ac6e8b566d9fe570

derpicknicker1, 2016-05-17:
Feature released in 1.3.0 
https://github.com/fabianonline/OctoPrint-Telegram/releases/tag/1.3.0

closed 2016-05-17 by derpicknicker1