jacopotediosi/OctoPrint-Telegram #22: Alternative to Pillow?

fabianonline, 2016-05-17:
Right now the plugin has Pillow as dependency - just to rotate the pictures from the webcam if necessary. But it is a really huge dependency taking a looooooong time to install on a Raspberry 1. Perhaps there is something more lightweight we could use for this task?

derpicknicker1, 2016-05-17:
Maybe PIL? I use this to resize the user images

fabianonline, 2016-05-17:
Pillow is a fork of PIL which has last been updated in 2009.

derpicknicker1, 2016-05-17:
Image manipulation and processing using Numpy and Scipy
http://www.scipy-lectures.org/advanced/image_processing/#geometrical-transformations

shaiss, 2016-05-20:
Octoprint already has an option to rotate the webcam.  Could that not be used?

fabianonline, 2016-05-21:
No. Octoprint just tells the browser to rotate the video stream. We can't do that with telegram. Although... perhaps we could just set that EXIF flag that is also used by digital cameras or smartphones to indicate the rotation of the image...?

derpicknicker1, 2016-12-29:
See also #16

jacopotediosi, 2025-07-20:
I couldn’t find any alternatives to Pillow that are significantly lighter and still allow easy handling of multiple formats (at least JPG and PNG, since there’s no guarantee about the format of the images coming from webcams).

Additionally, I noticed that some of the most popular OctoPrint plugins rely on Pillow (e.g., OctoApp, SpoolManager, Octolapse, OctoEverywhere, DiscordRemote, Octoslack, etc.).

The [OctoPrint download page](https://octoprint.org/download/#:~:text=Recommended%20hardware%3A%20Raspberry%20Pi%203B%2C%203B%2B%2C%204B%20or%20Zero%202.%20Expect%20print%20artifacts%20and%20long%20loading%20times%20with%20other%20options%2C%20especially%20when%20adding%20a%20webcam%20or%20installing%20third%20party%20plugins.%20Setups%20not%20using%20recommended%20hardware%20are%20not%20officially%20supported.) states that the recommended devices are Raspberry Pi 3B, 3B+, 4B, or Zero 2. On the Zero 2 — the least powerful of those — Pillow compiles in under 10 minutes. Also, I expect precompiled wheels for Pillow to be available for the most common architectures, avoiding local compilation entirely.

I'm closing this issue because I don’t believe this is a problem anymore with modern hardware. At this point, even obtaining a legacy device like a Raspberry Pi 1 is probably harder than upgrading.

closed 2025-07-20 by jacopotediosi