jacopotediosi/OctoPrint-Telegram #25: Error while updating OctoPrint-Telegram

MichMich, 2016-05-20:
While trying to upgrade OctoPrint-Telegram from version 1.1.1 to 1.3.1 I get the following error:

> Update failed!
> The update did not finish successfully. Please consult the log for details.

The log contains the following error:

```
2016-05-20 20:06:50,088 - octoprint.plugins.softwareupdate - ERROR - Update of telegram can not be performed
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 637, in _perform_update
    update_result = updater.perform_update(target, check, target_version)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/softwareupdate/updaters/pip.py", line 50, in perform_update
    install_arg = check["pip"].format(target_version=target_version)
KeyError: 'target'
```

I'm running OctoPrint Version: 1.2.11 (master branch)

derpicknicker1, 2016-05-20:
Sadly all versions lower than 1.3.0 will fail to update automatically. Please See issue #23 for more details.  Wenn are very sorry about that :heart:

closed 2016-05-20 by derpicknicker1

MichMich, 2016-05-21:
That fixed it. Thanks!