jacopotediosi/OctoPrint-Telegram #30: After upgrading - You are not allowed to do this! 😬

r0ndL, 2016-07-01:
After I upgraded Telegram.  I'm receiving "You are not allowed to do this! 😬" alert whenever I try to send a command.  In octiprint the Chat user has the correct rights.

daltonswayze, 2016-07-15:
Are you still having this issue? I have the same problem only I never had it working, just did a fresh install of octoprint 1.2.13 and follow the directions here https://github.com/fabianonline/OctoPrint-Telegram/blob/stable/README.md

flow23, 2016-07-17:
Dito

Xoius, 2016-09-17:
Yes, this is a little bit anoying ^^

orangeudav, 2016-09-17:
You must enable separate permissions _inside_ command and notify icons (click both on them) for the first time (and may be after upgrade)

Xoius, 2016-09-17:
That didn´t helped. I gave me the permission to send commands, saved it and rebooted but if i type a command in telegram i just get the response: You are not allowed to do this

Xoius, 2016-09-17:
Oh sry i figured it out.
But thanks for your fast response :D

closed 2016-09-24 by derpicknicker1

raphmeelo, 2017-05-11:
what it was?

victor-vct, 2017-06-02:
I'm having the same problem. All permissions are right, but the message "You are not allowed to do this" keep showing. What can it be?

derpicknicker1, 2017-06-07:
Did you follow this step from the README?

IMPORTANT
a) By clicking the pencil icon, you are able to enable/disable commands and notifications for the user/group in general. This will not affect the settings you made in 1. and 2.. It's an easy way to quickly activate/deactivate a user/group withot changing the settings. You will find the following options:

prothinker, 2020-09-06:
I am admin of telegram group and I have all permissions but when I am trying to add new admin it shows you are not allowed to do this🙁

heisgolden, 2021-07-30:
How did you solve this issue got a similar issue