jacopotediosi/OctoPrint-Telegram #32: notify on pause and resume

MarcusWolschon, 2016-07-11:
The plugin should also allow the user to get a notification on pause and on resume.

MiguelAngelLV, 2016-07-15:
Will it notifiy when I need change the filament?

MarcusWolschon, 2016-07-18:
When you need to change filament, https://github.com/MarcusWolschon/UltimateFilamentSensor/ could  pause the print and this plugin would then inform you about the pause.

Octoprint-Telegram has no knowledge about how much filament you have left and that's not the job of this plugin.

MiguelAngelLV, 2016-07-18:
I wanted to say when I used m600 code to change filament (for change color
filament). The printer pause the printing and wait.

I will want that, when the printer pause, it send my a message

MarcusWolschon, 2016-07-20:
Ok, my patch does a Telegram-notification when Octoprint does a pause.

M600 sounds like a g-code command that makes the printer wait for you to change filament.
As far as Octoprint is concerned, that waitins is part of printing as afterwards more commands will be sent.

This sounds like a different issue and has nothing to do with the Telegram-Plugin or this specific feature-requres for the Telegram-Plugin.

derpicknicker1, 2016-09-23:
Hey. I was quite busy the last months. Started a new job...so i had no time to work on the plugin.
Thank you @MarcusWolschon for your pull request. I'm not sure if i will integrate it. My plan is to let the user define own notification hooks on every event octoprint throws. But it will take a bit of time to integrate this. Especially the settings system (care of the the rights of users) will take some work. I think this dynamic way of adding notifications will be a better way than to hardcode more notifications. 
This topic is one of the most top items on my list. So stay tuned :)

derpicknicker1, 2016-09-29:
Done in 1.4.0 (will come soon).

closed 2016-09-29 by derpicknicker1

AdmiralHawkins, 2019-01-03:
Hi, derpicknicker1, I installed your plugin, it sends me notification when print starts and when it ends but not when a m600 command occurs, even if I enabled notifications for pause event. Do I have to configure/enable something in particular to catch m600 command?