jacopotediosi/OctoPrint-Telegram #34: Send webcam captured imaged even if the printer is not connected

svh1985, 2016-07-11:
I would love too also receive a picture if I send the /status command, even if the printer is disconnected.
Usually, at least in my case, the webcam remains connected to the Octoprint server and pointed to the 3D printer even if the printer is offline.

Thanks!

MarcusWolschon, 2016-08-01:
I can't figure it out. Sombody else should implement this.

derpicknicker1, 2016-10-12:
Done in [1.4.0](https://github.com/fabianonline/OctoPrint-Telegram/releases/tag/1.4.0)

DaAwesomeP, 2017-10-12:
I'm having issues with this. If the printer is not connected and I send `/status`, I just get (without an image):
```
⚠️ Not connected to a printer. Use /con to connect.
```

DaAwesomeP, 2017-10-12:
I see there's a related TODO here on line 77: https://github.com/fabianonline/OctoPrint-Telegram/blob/c04fec95b28f32d57b3adf47ba00354639d9ac32/dev_utils/testBot/telegramCommands.py#L74-L83

DaAwesomeP, 2017-10-12:
Is it possibly because there is not an `'image': True` setting here?
https://github.com/fabianonline/OctoPrint-Telegram/blob/c04fec95b28f32d57b3adf47ba00354639d9ac32/octoprint_telegram/telegramNotifications.py#L69-L72
I would love to submit a pull if this is a fix!

derpicknicker1, 2018-01-08:
I will review this