jacopotediosi/OctoPrint-Telegram #35: user can't modify /status message

MarcusWolschon, 2016-07-11:
The user can modify all kinds of messages in the Octoprint settings dialog.
The status when not printing is there.
The status when printing is missing in the dialog.

derpicknicker1, 2016-09-23:
It is the same text as defined for z-change. Do you need a seperate text for "Statuss Printing " and z-Change?

MarcusWolschon, 2016-09-27:
Maybe it's enough to change the wording to make it obvious that "start printing" is a special case of a "z-change" in the settings dialog.

derpicknicker1, 2016-10-01:
There allready is a hint in the stettings dialog.

[image]

derpicknicker1, 2016-10-21:
I think this can be closed.

closed 2016-10-21 by derpicknicker1