jacopotediosi/OctoPrint-Telegram #36: add /resume and /pause commands

MarcusWolschon, 2016-07-12:
It seems that I can start as print but not **/resume** it.
Neither can I **/pause** it, when I detect something is odd and want to have the printer wait until I come back to fix it without loosing the half-finished print.

MarcusWolschon, 2016-07-12:
Looks easy enough to add in
https://github.com/fabianonline/OctoPrint-Telegram/blob/master/octoprint_telegram/telegramCommands.py

according to
http://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface.toggle_pause_print
there is a similar 
self.main._printer.toggle_pause_print()
command we can use.

MarcusWolschon, 2016-07-12:
untested code:
https://github.com/MarcusWolschon/OctoPrint-Telegram/commit/9907a56b495e25846ebe3fe9668102989afdb3e7

derpicknicker1, 2016-09-29:
Done in 1.4.0 (will come soon).

closed 2016-09-29 by derpicknicker1