jacopotediosi/OctoPrint-Telegram #37: Unable to save settings for OctoPrint-Telegram

Fetto, 2016-07-18:
I have been unable to save changes in settings for this plugin since after I upgraded for the first time (the first installation was ok). The only thing that I can save is the chat id, nothing else gets saved (can't even enable debugging). Tried uninstalling and removing everything from the config-file, still same problem.
I can't get past the "save first or trashcan-click".

Running version 1.3.0.dev1245+gcba742a (devel branch).

Fetto, 2016-08-06:
2016-08-06 17:26:52,300 - octoprint.server.api.settings - ERROR - Could not save settings for plugin Telegram Notifications (1.3.2)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev1286_g0394121-py2.7.egg/octoprint/server/api/settings.py", line 319, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 712, in on_settings_save
    if data['chats']:
KeyError: 'chats'

joris57, 2016-09-09:
I am experiencing the same issue in a first-time install..

joris57, 2016-09-11:
Update: I installed the master branch of Telegram and the 'chats' error message has now been replaced by 'octoprint.plugins.telegram.listener - ERROR - Exception caught! global name 't' is not defined'.

Same result: settings are not saved.

I am on OctoPrint: 1.3.0.dev1288+gedf83e4 and Telegram Notifications: 1.3.2 (master branch).

derpicknicker1, 2016-09-24:
It seems like this is a bug only in devel branch. At the moment there is only time to care for a running version for the offical releases.
I will look on this again when octoprint 1.3.0 releases.

derpicknicker1, 2016-09-24:
Okay. I had another look at it, and ther actually was a bug. Will be fixed with next update.

derpicknicker1, 2016-09-24:
Fixed in https://github.com/fabianonline/OctoPrint-Telegram/releases/tag/1.3.3

closed 2016-09-24 by derpicknicker1