jacopotediosi/OctoPrint-Telegram #38: Plugin install related to Telegram

Crunklin, 2016-08-24:
**Running OctoPi on a Raspberry Pi 3**

When trying to install the telegram plugin I am receiving the following error message. I believe I am following the correct procedure and have not found any helpful info in either this forum or google. 

Please help...

pi@octopi:~/oprint/bin $ pip install https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip
Downloading/unpacking https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip
  Downloading stable.zip (85kB): 85kB downloaded
  Running setup.py (path:/tmp/pip-0tJZ6P-build/setup.py) egg_info for package from https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
    Complete output from command python setup.py egg_info:
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 255 in /tmp/pip-0tJZ6P-build
Storing debug log for failure in /home/pi/.pip/pip.log

derpicknicker1, 2016-09-24:
I can not reproduce this behavior. Also this looks more like a problem in your octoprint/system setup. Are you able to install other plugins?

closed 2016-09-24 by derpicknicker1

NikSays, 2018-07-17:
Get the same error, able to install other plugins

skallen, 2019-04-27:
How/Was this resolved?  I get the same error with a different plugin.