jacopotediosi/OctoPrint-Telegram #39: webcam image not sending even when requested

JimGaleForce, 2016-09-25:
not getting images (which work thru octoprint just fine, both video and snapshot). settings request images for /status, (start/stop/etc). none of them produce images.

here's the relevant debug info for the /status request:
(odd that there's a with_image=True, and a snapshot url followed by a  Sending without image):

2016-09-25 07:14:33,698 - octoprint.plugins.telegram.listener - INFO - Got a command: '/help' with parameter: '' in chat 295263708
2016-09-25 07:14:40,625 - octoprint.plugins.telegram.listener - INFO - Got a command: '/status' with parameter: '' in chat 295263708
2016-09-25 07:15:08,029 - octoprint.plugins.telegram.listener - INFO - Got a command: '/list' with parameter: '' in chat 295263708
2016-09-25 07:16:16,278 - octoprint.plugins.telegram.listener - INFO - Got a command: '/status' with parameter: '' in chat 295263708
2016-09-25 07:17:28,620 - octoprint.plugins.telegram.listener - INFO - Got a command: '/status' with parameter: '' in chat 295263708
2016-09-25 07:17:55,001 - octoprint.plugins.telegram.listener - INFO - Got a command: '/status' with parameter: '' in chat 295263708
2016-09-25 07:24:37,374 - octoprint.plugins.telegram.listener - INFO - Got a command: '/status' with parameter: '' in chat 295263708
2016-09-25 07:33:29,551 - octoprint.filemanager.analysis - INFO - Starting analysis of local:testblock.gcode
2016-09-25 07:38:37,489 - octoprint.plugins.telegram.listener - INFO - Got a command: '/status' with parameter: '' in chat 295263708
2016-09-25 07:44:39,310 - octoprint.plugins.telegram.listener - DEBUG - MESSAGE: {u'message': {u'from': {u'first_name': u'Jim', u'last_name': u'Gale', u'id': 295263708}, u'text': u'/status', u'entities': [{u'length': 7, u'type': u'bot_command', u'offset': 0}], u'chat': {u'first_name': u'Jim', u'last_name': u'Gale', u'type': u'private', u'id': 295263708}, u'date': 1474789479, u'message_id': 21}, u'update_id': 785575775}
2016-09-25 07:44:39,312 - octoprint.plugins.telegram.listener - INFO - Got a command: '/status' with parameter: '' in chat 295263708
2016-09-25 07:44:39,314 - octoprint.plugins.telegram - DEBUG - Got an event: StatusNotPrinting Payload: {}
2016-09-25 07:44:39,319 - octoprint.plugins.telegram.TMSG - DEBUG - Printer Status{'progress': {'completion': 100, 'printTime': 273, 'filepos': 7856L, 'printTimeLeft': 0}, 'state': {'text': 'Operational', 'flags': {'operational': True, 'paused': False, 'printing': False, 'sdReady': False, 'error': False, 'ready': True, 'closedOrError': False}}, 'offsets': {}, 'job': {'estimatedPrintTime': 32.14438007491633, 'filament': {'tool0': {'volume': 0.0, 'length': 1.5529000000000004}}, 'averagePrintTime': 273.83736085891724, 'file': {'origin': 'local', 'date': 1474788809, 'name': u'testblock.gcode', 'size': 7856L}, 'lastPrintTime': 273.83736085891724}, 'currentZ': 10.0}
2016-09-25 07:44:39,320 - octoprint.plugins.telegram.TMSG - DEBUG - TEMPS - {'bed': {'actual': 41.2, 'target': 0.0, 'offset': 0}, 'tool2': {'actual': 58.5, 'target': 0.0, 'offset': 0}, 'tool1': {'actual': 50.9, 'target': 0.0, 'offset': 0}, 'tool0': {'actual': 84.7, 'target': 0.0, 'offset': 0}}
2016-09-25 07:44:39,322 - octoprint.plugins.telegram.TMSG - DEBUG - Sending Notification: Not printing.
Bed 41.2/0.0, Extruder 84.7/0.0.
2016-09-25 07:44:39,324 - octoprint.plugins.telegram - DEBUG - Sending a message: Not printing.\nBed 41.2/0.0, Extruder 84.7/0.0. with_image=True chatID= 295263708
2016-09-25 07:44:39,326 - octoprint.plugins.telegram - DEBUG - Snapshot URL: /webcam/?action=snapshot
2016-09-25 07:44:39,331 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True, 'reply_markup': '{"hide_keyboard": true}'}
2016-09-25 07:44:39,332 - octoprint.plugins.telegram - DEBUG - Sending without image.. 295263708
2016-09-25 07:44:40,346 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
2016-09-25 07:44:40,352 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 785575776...
2016-09-25 07:44:51,850 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAa1aAAAaA1AA-AAaAAaAAAaA1aAAAA1A1
2016-09-25 07:45:11,217 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 785575776...
2016-09-25 07:45:42,125 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 785575776...

derpicknicker1, 2016-09-25:
It seems like the plugin can't take a picture. Is the snapshot-URL the one you get on fresh installs? Or did you change it? 
The plugin can't resolve a simple "/webcam/[...]" URL. There has to be a full address.
Set the snapshot URL to one of the following:
* http://127.0.0.1:8080/?action=snapshot
* http://127.0.0.1/webcam/?action=snapshot

Both should work. Without the full address the path is only useable from within the browser.

derpicknicker1, 2016-09-25:
**Memo**
We need full URL
Simply inject "http://127.0.0.1" if leading '/' in URL?
Py sockets to determine local IP?

JimGaleForce, 2016-09-25:
that worked to get a photo. however, wouldn't that fail for remote use as it would be relative to the browser. If so, this is a partial solution.

and yes, installed from the [INSTALL] button after finding it from the repo.

thanks!

derpicknicker1, 2016-09-25:
Octoprint uses the stream url. For me there is nothing broken in octoprint web ui. Does octoprint, or any component of it, even use the snapshot url?

JimGaleForce, 2016-09-25:
yes, at least the timelapse feature

derpicknicker1, 2016-09-25:
When you checked your print by snapshot, how did you doch that? Typing a url to your browser? Custom button in Octoprint?

JimGaleForce, 2016-09-25:
http://(my home ip address):(my forwarded port)/webcam?action=snapshot
... and just tried it and it works ... so since I'm not requesting octoprint to deliver it for me - no issue.
(shall I Close the issue or you? :) ) - thanks!

derpicknicker1, 2016-09-25:
I'm glad that it works for you now. So i close this issue now.

closed 2016-09-25 by derpicknicker1

macscripter, 2016-10-09:
I have the same problem... I can take snapshot on my local url:  http://192.168.1.12/webcam/?action=snapshot

But I  don't receive the message from the boot Telegram with the photo, only text... Could someone help?

Thanks in advance

derpicknicker1, 2016-10-11:
As i worte before: open settings in octoprint  and select webcam. For snapshot url type in `http://127.0.0.1/webcam/?action=snapshot`

superJT1, 2019-04-21:
I have an IP Cam (Model: skytronic 351.149), which is not directly wired to the computer (Beaglebone Black) and it works perfectly directly in Octoprint, but it doesn't work in Telegram. I always get the notification "[ERR GET IMAGE]". The webcam-link that I use in Octoprint has this style:
 
"http://1.1.1.1:11/snapshot.cgi?user=USER&pwd=PWD"

I think I am already in snapshot-mode (because a "snapshot.cgi" is part of the URL), and adding "/webcam/?action=snapshot" doesn't work here. I tried adding this to the end of the URL but this changes nothing - it still works directly in Octoprint, but doesn't work in Telegram. Do you have any idea what to do?

juampick, 2019-08-13:
I'm having the same issue @superJT1 using IP Cam software for ios and using an iphone camera. The camera works for octoprint but not on octoprint-telegram plugin.