jacopotediosi/OctoPrint-Telegram #41: add /feedrate to comand list

Raffa2s, 2016-09-28:
can you add /feedrate to comand list? somenthing like 10% 25% 30% 40% 50% ecc trought bottons or trought input by keyboard.
thanks a lot

derpicknicker1, 2016-09-29:
It's on my to do list for next release.
Meanwhile you may define some [custom controls ](http://docs.octoprint.org/en/master/features/custom_controls.html#sec-features-custom-controls) or do this via a [plugin](http://plugins.octoprint.org/plugins/customControl/).

You can use them by sending `/ctrl` to the bot. But only static controls (without input/output) are useable until now (i'm also working on this). So you have to define one for 10%, one for 25% and so on.

derpicknicker1, 2016-10-12:
Done in [1.4.0](https://github.com/fabianonline/OctoPrint-Telegram/releases/tag/1.4.0)

closed 2016-10-12 by derpicknicker1