jacopotediosi/OctoPrint-Telegram #42: Laer count

krwynn, 2016-10-02:
Hi. Have this plugin installed and it is working great. Is there any way that as a part of the status message it can report what layer its on? Such as layer # of #. Thanks.

derpicknicker1, 2016-10-03:
Since octoprint only provides current z height, this would be a little tricky to implement. May it can be calculated by layerheight. But that height has to be set for every print. 
Only the GcodeViewer in the browser knows things like layer # and layer height. But that is not on server side, so the server can't know these values.
Same problem as in #8

krwynn, 2016-10-06:
During printing, on the Gcode viewer tab in octoprint, it reports out the layer#. Maybe there's a way to grab that variable? I'm guessing as I have no idea.

derpicknicker1, 2016-10-08:
This is only available on client site (in browser). Everything the gecode viewer shows/does happens in your browser, not in octoprint.

krwynn, 2016-10-09:
Ok. Thank you.

closed 2021-06-14 by giloser