jacopotediosi/OctoPrint-Telegram #50: Edit user Command or Notify

temuccio, 2016-10-31:
Hello, 
when I set authorization on one user pressing pencil, 
I have this error (on inspect element of Chrome):
Uncaught TypeError: Unable to process binding "checked: function (){return settings.settings.plugins.telegram.chats['269074525']
Do you can help me

derpicknicker1, 2016-10-31:
Are you using chrome?
This seems to be a problem with chrome caching the settings data. Try to clear your website cache and relaod the page (while you are logged in). I try to solve it. Also i will inform Gina (Dev of Octoprint) because i think it is a problem within octoprint.

derpicknicker1, 2016-10-31:
I reportet this potential bug(https://github.com/foosel/OctoPrint/issues/1567)
Maybe you can test in other browsers? For me it worked wit IE.

commit b451c0e7 (2016-12-13): Merge pull request #52 from foosel/fix/chatsMustBeDict

`chats` must be provided as dict on settings
Should fix #50 and #54
closed 2016-12-13 by derpicknicker1