jacopotediosi/OctoPrint-Telegram #51: Especify for each print to who to notify

angelocarlotto, 2016-11-09:
Dude, here is a great feature to your grate plugin would be the ability to specify for each printing file or folder(octoprint 1.3.0rc) to whom send notifications.

This feature will help to notify clients the status of their printing files or folders in case they have more then one file to print.

angelocarlotto, 2016-11-09:
Could be in the configuration chats panel. The ability to specify all prints or only prints files or print folder.

derpicknicker1, 2016-11-09:
I also thought about this feature. I don't know when i will have time for this. I will put it on my list. But first i have to solve some problems with 1.3.0 in general, before 1.3.0 will be released.

angelocarlotto, 2016-11-10:
Thank for the attention, man.

I am a .NET developer. If you could give some direction may i could help you i little thing at the beginning.

angelocarlotto, 2016-11-10:
Since you mentioned about problem with 1.3.0 version. There is any ticket about the command /files versus the folder feature that there is in version 1.3.0?

If i have files inside folders, i don't know how, but just the files in the root directory are listed.

Hope you can reproduce this issue.

And Man, congrats for this plugin. Its awesome. Better then other dedicate mobiles apps i have ever try. It was grate the ideia to use a platform already build , already tested just like telegram and integrate with octoprint another well know platform.

derpicknicker1, 2016-11-10:
Thank you!
As you can see in #45 the folder support is planned for the new version. Until then you can only view the files in the root folder because this is the state of the art in offical stable version of octoprint.