jacopotediosi/OctoPrint-Telegram #56: Settings are not saved/changed

JosvdP, 2016-12-08:
I run the latest version of the Telegram plugin (1.4.0) on the latest version of Octoprint (1.3.0) and I can't save/change my settings for some reason.

I've managed to add my Telegram Token and save the settings but now it doesn't accept anymore changes to the Telegram settings.

I've tried to reinstall the plugin but it remembered my old token settings so it wasn't fully uninstalled? Is this an issue with permissions on the settings file?

JosvdP, 2016-12-08:
Luckily I was smart enough to opt-in for the logging errors in main octoprint log and I found the following info in the log:

```
2016-12-08 21:37:25,602 - octoprint.server.api.settings - ERROR - Could not save settings for plugin Telegram Notifications (1.4.0)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/server/api/settings.py", line 386, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 722, in on_settings_save
    if data['chats']:
KeyError: 'chats'
2016-12-08 21:37:47,788 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/util/__init__.py:562: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if source == target:

2016-12-08 21:37:47,790 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/util/__init__.py:580: UnicodeWarning: Unicode unequal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if value_source != value_target:

2016-12-08 21:43:16,164 - octoprint.server.api.settings - ERROR - Could not save settings for plugin Telegram Notifications (1.4.0)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/server/api/settings.py", line 386, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 722, in on_settings_save
    if data['chats']:
KeyError: 'chats'
```

andatovid, 2016-12-08:
Hello I have the same problem. I can't save the settings.

2016-12-08 23:01:57,263 - octoprint.server.api.settings - ERROR - Could not save settings for plugin Telegram Notifications (1.4.0)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/server/api/settings.py", line 386, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 722, in on_settings_save
    if data['chats']:
KeyError: 'chats'

foosel, 2016-12-09:
Probably a duplicate of #44. Sent a PR to fix that (see #53).

well-engineered, 2016-12-11:
I have this problem as well, since updating octoprint.

saxos1983, 2016-12-11:
I patched the code with #44 and the problem is solved so far. I hope they will review the pull request and merge it into the code.

JosvdP, 2016-12-12:
I patched the code with #44 as well but I cannot edit a user settings by clicking the pencil icon. Nothing seems to happens when I click it.

closed 2016-12-13 by derpicknicker1

SillyHats, 2018-07-09:
I am having this same issue with OctoPrint 1.3.8 and Telegram Notifications 1.4.2
Something seems to have broken with since #53 

`2018-07-09 16:37:36,865 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1200.0
|    ram: 918192128
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    octopi_support:
|      model: 3B
|      revision: a22082
|      version: 0.15.1
|  python:
|    pip: 9.0.3
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
2018-07-09 16:37:36,876 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2018-07-09 16:37:36,894 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2018-07-09 16:37:37,124 - octoprint.server - INFO - Shutting down intermediary server...
2018-07-09 16:37:37,453 - octoprint.server - INFO - Intermediary server shut down
2018-07-09 16:37:37,456 - octoprint.events - INFO - Processing startup event, this is our first event
2018-07-09 16:37:37,489 - octoprint.events - INFO - Adding 1 events to queue that were held back before startup event
2018-07-09 16:37:37,509 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2018-07-09 16:37:37,652 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2018-07-09 16:37:37,930 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2018-07-09 16:37:37,933 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2018-07-09 16:37:37,951 - octoprint.util.connectivity_checker - INFO - Connectivity changed from online to offline
2018-07-09 16:37:38,859 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "CR-10" for _http._tcp
2018-07-09 16:37:38,881 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "CR-10" for _octoprint._tcp
2018-07-09 16:37:38,902 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2018-07-09 16:37:39,155 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2018-07-09 16:37:39,212 - octoprint.plugins.roomtemp - INFO - This is not a Raspberry Pi - Plugin halted
2018-07-09 16:37:40,079 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2018-07-09 16:37:41,253 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2018-07-09 16:37:41,441 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2018-07-09 16:37:41,889 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2018-07-09 16:37:41,948 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2018-07-09 16:37:42,371 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2018-07-09 16:37:43,011 - octoprint.server.util.sockjs - INFO - New connection from client: 127.0.0.1
2018-07-09 16:37:43,514 - octoprint - ERROR - Exception on /api/plugin/telegram [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 858, in on_api_get
    'bind_cmd':[k for k, v in self.tcmd.commandDict.iteritems() if 'bind_none' not in v ],
AttributeError: 'NoneType' object has no attribute 'commandDict'
2018-07-09 16:37:43,529 - tornado.access - ERROR - 500 GET /api/plugin/telegram?bindings=true (127.0.0.1) 25.75ms
2018-07-09 16:37:46,549 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 12.85ms
2018-07-09 16:37:46,619 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 23.32ms
2018-07-09 16:37:46,639 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 17.10ms
2018-07-09 16:37:46,703 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 20.75ms
2018-07-09 16:37:46,722 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 15.83ms
2018-07-09 16:37:46,774 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 14.59ms
2018-07-09 16:37:46,790 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 13.44ms
2018-07-09 16:37:46,847 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 13.40ms
2018-07-09 16:37:46,865 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 15.68ms
[... 495 lines cut ...]
[... cut ...]
2018-07-09 16:56:34,799 - octoprint.plugins.telegram - DEBUG - Exception: 'NoneType' object has no attribute 'msgCmdDict'
2018-07-09 16:56:34,994 - octoprint.plugins.telegram - DEBUG - Exception: 'NoneType' object has no attribute 'msgCmdDict'
2018-07-09 16:56:43,248 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2018-07-09 16:57:13,916 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2018-07-09 16:57:44,565 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...`

commit ebfc1208 (2016-12-13): Merge pull request #53 from foosel/fix/issue44

Fix for issue #44 (new settings behaviour in OctoPrint 1.3.0)
Fix #44 
Fix #56