jacopotediosi/OctoPrint-Telegram #57: Webcam image has the wrong orientation

kevinrombach, 2016-12-10:
The image sent from the bot has another orientation like the stream in the control tab when i change the orientation of the webcam in the octoprint settings. How can i fix that?

derpicknicker1, 2016-12-13:
The plugin uses the settings you make for the webcam in Octoprint. So the image should be rotated/flipped as you set it in the webcam settings. I can not reproduce this issue.

derpicknicker1, 2016-12-30:
This should not happen. I will check this again.

derpicknicker1, 2017-01-03:
I found the issue. Telegram bot rotates the images 90° clockwise. OctoPrint does this counter clockwise (as the option in webcam settings tell). Will be fixed in next update.

derpicknicker1, 2017-01-16:
Still some issues :disappointed:

closed 2017-01-27 by derpicknicker1

NiklasOlofsson, 2017-04-15:
Getting upside down picture in Telegram with the latest release. I have "Rotate webcam 90 degrees counter clockwise" enabled.

Sunoo, 2017-07-20:
I'm also getting upside down pictures in Telegram with the camera rotated 90 degrees in Octoprint.

OdomLinx, 2017-10-27:
I have the same problem too.

kaju666, 2018-01-07:
Any updates, problem still exist.

Guduche, 2018-02-12:
I have the same problem too

GilDev, 2019-06-01:
Same problem… Image flipped 180°… ☹️

carlosgalvezp, 2019-08-15:
Same problem here with latest release. I had to spend quite some time playing with the different rotations in Octoprint to end up getting a straight image in Telegram. But of course now the image in Octoprint is flipped...

Vabe7, 2019-09-11:
I also have this issue, the image is flipped...

giloser, 2019-09-11:
Hi,
to be sure I understand this correctly you didn't change the rotation in the \boot\octopi.txt file but in the interface of Octoprint? And then it's not correct between octoprint and telegram?
I change my config in the file so I flipped the image before octoprint create the stream and I don't have to change it to have it correct in telegram.

th3darkang3l, 2020-03-06:
I found a workaround and fixed locally this problem... the issue is that whatever rotation CW or CCW you have the plugin do always the same rotation (this line of code https://github.com/fabianonline/OctoPrint-Telegram/blob/b1f90950293575005d3be4d791602a662222a366/octoprint_telegram/__init__.py#L1321 )

I beg some collaborator to fix the issues managing the two types of rotation, in the meantime to fix the issues locally you can : 
- putty/ssh into your box
- nano/vi /oprint/lib/python2.7/site-packages/octoprint_telegram/__init__.py
- find ROTATE_270 and change to ROTATE_90
- save and reboot
Now images that are sent to your telegram bot are no longer upside down.

danst0, 2021-01-30:
Still open

axelschmidt91, 2021-02-19:
I get the same problem. Is it possible to reopen this issue?

giloser, 2021-02-19:
Like I said on the #287 I'll will on the next release add an option to rotate on 270 or 90° so you could choose the best for you.

alakond, 2025-01-11:
> I found a workaround and fixed locally this problem... the issue is that whatever rotation CW or CCW you have the plugin do always the same rotation (this line of code [...]

Didn't help. Doesn't work. Any other ideas?

commit a6ccf5fc (2017-01-03): Fix issue: worng image rotation direction

* Telegram plugin rotates the images 90° clockwise. OctoPrint does this
counter clockwise. So plugin is now fixed and will also rotate 90°
counter clockwise.
* Fix #57
commit 77ce709f (2017-01-27): Fix issue on image orientatation

Fix #57