jacopotediosi/OctoPrint-Telegram #58: Add slice command

derpicknicker1, 2016-12-13:
It would be great to have the ability to slice right from telegram. I have the idea of adding an SLICE command to upload an STL file and slice by choosing a cura_engine profile. Then save the generated gcode for printing. (Idea by @garzadba)

cnelsonsic, 2017-04-15:
Has anyone worked on this yet? I might want it bad enough to implement it.

derpicknicker1, 2017-06-07:
Still working on this. I have not much time for plugin development at the moment. But it will come. Stay tuned.

jacopotediosi, 2025-08-17:
After the release of the upcoming version 1.9.0, I will work on this feature, with the current plan to include it in version 1.10.0.

A few notes:

OctoPrint includes hooks that allow plugins to register themselves as slicers.

At the moment, the plugins I found that provide slicer functionality are:
- [CuraEngine Legacy](https://plugins.octoprint.org/plugins/curalegacy/)
- [Full-featured Slicer](https://plugins.octoprint.org/plugins/slicer/)
- [InternalSlicer](https://github.com/Garr-Garr/OctoPrint-InternalSlicer)
- [PrePrintService](https://plugins.octoprint.org/plugins/preprintservice/)
- [Slic3r](https://plugins.octoprint.org/plugins/slic3r/)

OctoPrint-Telegram will be able to query OctoPrint for available slicers, configured slicing profiles, and perform slicing operations on files through the APIs under the `/api/slicing/` path, as defined in [the documentation](https://github.com/OctoPrint/OctoPrint/blob/89733fd12fbd1b4ad9be08ee1f4ccafefdafe340/docs/api/slicing.rst).

The addition of this feature will involve:
- creating a new `/slice` command, with its dedicated permission, and adding the command to the [Wiki](https://github.com/jacopotediosi/OctoPrint-Telegram/wiki).
- enabling the possibility to also display STL files through the `/files` command. This will require a review of all file-related functions, since they may no longer be limited to machinecode files. For example, some details (such as integration with the [Cost Estimator](https://plugins.octoprint.org/plugins/cost/) plugin) will need to be excluded when displaying details of STL files.
- showing a "slice" button in messages related to files, if the selected file is an STL and the user has permission to use the `/slice` command.
- introducing a new default emoji named `slice`.

jacopotediosi, 2025-09-18:
Finally implemented:
[image]

closed 2025-09-18 by jacopotediosi