jacopotediosi/OctoPrint-Telegram #61: Installation error

joselbachi, 2016-12-29:
Hi,

When i try to install the plug-in I get this error:

pi@octopi:~/oprint/bin $ ./pip install https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip
Collecting https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip
  Downloading https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip
     \ 102kB 219kB/s
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg (from OctoPrint-Telegram==1.4.1)
Collecting pillow (from OctoPrint-Telegram==1.4.1)
  Downloading Pillow-3.4.2.tar.gz (10.8MB)
    99% |████████████████████████████████| 10.8MB 696kB/s eta 0:00:01Exception:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 324, in run
    requirement_set.prepare_files(finder)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 882, in _download_http_url
    _download_url(resp, link, content_file, hashes)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 603, in _download_url
    hashes.check_against_chunks(downloaded_chunks)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/hashes.py",  line 46, in check_against_chunks
    for chunk in chunks:
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 571, in written_chunks
    for chunk in chunks:
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/ui.py", line 139, in iter
    for x in it:
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 560, in resp_read
    decode_content=False):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/response.py", line 357, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/response.py", line 314, in read
    data = self._fp.read(amt)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/filewrapper.py", line 63, in read
    self._close()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/filewrapper.py", line 50, in _close
    self.__callback(self.__buf.getvalue())
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/controller.py", line 275, in cache_response
    self.serializer.dumps(request, response, body=body),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/serialize.py", line 87, in dumps
    ).encode("utf8"),
MemoryError

And I can't install Pillow manually.

Sorry for my English.

derpicknicker1, 2016-12-29:
This may really is an issue. We want to go away from pillow, but have no real alternative. May you have a suggestion?
See also issue #22

joselbachi, 2016-12-29:
I now nothign about Phyton, but it cloud be:

https://wiki.python.org/moin/PythonMagick or https://wiki.python.org/moin/ImageMagick

derpicknicker1, 2016-12-29:
I will close this now because it seems that you know the failure. I left a reference in #22 to this issue.

closed 2016-12-29 by derpicknicker1