jacopotediosi/OctoPrint-Telegram #64: Token status is: Token valid for UNKNOWN.

cyranoidy, 2017-01-19:
Hello,
First of all thank you for this excellent plugin, i have been using it for a couple of days now without problem, till now, 

for your info, i am running:

octoprint Version: 1.3.0.post1+g1014712 (master branch)

telegram version : Telegram Notifications (1.4.2)

octopi image flashed: 2017-01-18_2016-11-25-octopi-jessie-lite-0.14.0.zip

today i did two things and after that my telegram stopped working, first one, i connectted the pi wo a ethernet cabel, before it was on wifi, and i modified the settings of wifi network in the ssd of the pi, i erased the lane of the ssid ad the password of the wifi, and i activated the one for assigning an IP to the ethernet cable plugged in,

(seemed that the pi ignored all my commands, as now i can acces the octoprint interface by 2 different IP's i think it created an ip adress for the wifi conn and an adress for the eth. but anyway that is something i will work on...

the other think i did is overclocking the Pi, i used this commands:

arm_freq=1350
core_freq=450
over_voltage=4
disable_splash=1
force_turbo=1 #6 (uncomment to avoid CPU scaling down to 600Mhz)
boot_delay=1 #helps to avoid sdcard corruption when force_turbo is enabled.
sdram_freq=500 #57  Works only with some boards.
gpu_mem=250

so after all this changes, i noticed that the pi stopped sending telegram messages, i test the token and it says: Token status is: Token valid for UNKNOWN.  but doesent appear the name of my bot,

any ideas how to solve the issue?

Thanks for your support guys!

Regards,

derpicknicker1, 2017-01-20:
That is strange. Two things:
1. Please activate debugging in telegram plugin settings (OctoPrint-Settings -> Telegram -> scroll down to the end). Then restart octoprint and try to connect your bot and let the bot send/receive messages. Next take the octoprint.log file, create a [gist](https://gist.github.com) and post the url here.
2. Create a new bot ([@Botfather](http://telegram.me/botfahter) and try to setup the plugin with the new bot token. I experienced also some issues with bots and got them only solved by creating a new one.

cyranoidy, 2017-01-20:
Well i got to say i have it solved, looks like the pi did not have connection to the internet, i deactivated the lines in octopi-network.txt

Iface eth0:1 inet static
Adress 192.168.1.178
Netmask 255.255.255.0
Broadcast 192.168.1.1

And after 2 reboots looks that its back to life again

Thank god.

I will keep in mind the debugging option for future issues 

Thanks a million Derpick for your answer , i hope if  someone gets this error knows now a little bit more what can cause it.

Have a nice weekend!!

derpicknicker1, 2017-01-20:
I'm glad to hear that. Have a nice weekend too.

closed 2017-01-20 by derpicknicker1