jacopotediosi/OctoPrint-Telegram #70: Custom system commands not working

viljamip, 2017-03-08:
I added System commands to turn my printer on and off. They work great when I use them with my browser on the Octoprint UI. When I use the /sys command on Telegram, the bot replies back showing my custom system commands but the buttons do not work. 

The normal Restart Octoprint etc. work normally but when I press on my "Turn printer on" button, Telegram will pop up a banner that says Loading... and nothing else happens.

I'm using latest version of Octoprint and this plugin. Am I missing something or is this a bug?

viljamip, 2017-03-10:
I made my System actions with the "System Command editor" plugin and they worked with Telegram. The ones I had before I created manually by editing the config.yaml. Those worked with Octoprint UI but not with Telegram.

closed 2017-03-10 by viljamip

derpicknicker1, 2017-03-12:
May you please post your entries yo made manualli in config.yaml, so i can have a look what may went wrong?

viljamip, 2017-03-12:
I found this from a backup and tested that it behaved like I described. I censored API keys and other IDs with XXX. Hopefully this helps.
[https://dl.dropboxusercontent.com/u/5129261/config-xxx.yaml](url)