jacopotediosi/OctoPrint-Telegram #72: Flip webcam doesn't have effect

CEG-LTU, 2017-03-16:
Hello,

I have the following problem, I set up a webcam with my OctoPi and I had to rotate it and flip it vertically and horizontally. On the web, server image looks correct, but on telegram is upside down. It looks like it is only doing rotation 90 degrees but doesn't flip an image. 

Is there any solution for such problem?

jkronen, 2017-05-01:
Why is this closed? I have the same problem

derpicknicker1, 2017-06-06:
Solution will come. I have to do an update but dont have time for it. Sorry. Stay tuned.

userosos, 2017-11-29:
It not work now, please update! How we can make it plugin from source? Does anyone have an instruction for make it from source?

kaju666, 2018-01-14:
Any bugfix for that?

b0b0x, 2018-01-23:
Hi, same issue... could you please fix it?

The plugin is rotating clockwise instead of counterclockwise setting in octoprint, Then the telegram picture is always upside down compared to the webcam image in octoprint...

kaju666, 2018-01-23:
@b0b0x i started using octoslack, and its very good for notofications

rlogiacco, 2019-07-17:
Let's merge the discussion on issue #146 : we have applied a pull request which should have this fixed, but we need feedback

closed 2019-07-17 by rlogiacco