jacopotediosi/OctoPrint-Telegram #73: Error: 'NoneType' object has no attribute 'username'

ido1990, 2017-03-16:
Hi!
I installed the plugin couple days ago and it worked.
Suddenly it stopped and shows an error:
Error: 'NoneType' object has no attribute 'username'
Screenshot:
[https://i.imgsafe.org/afaae87715.png](https://i.imgsafe.org/afaae87715.png)

Tried to disable and enable the plugin but it didn't solve it.

Thanks!

PaulsPi, 2017-03-20:
Hey, i got the same problem! Did you already solve it?

ido1990, 2017-03-20:
No... 
I changed the token from the botfather and it temporarily fixed but it's back again. Also the bot stopped responding even if the token is working.. .

PaulsPi, 2017-03-23:
I guess it has something to do with the TouchUI plugin... I've been deleting the whole raspberry and booting it again, installing telegram and many other plugins. After installing the TouchUI plugin and the autostart of it the telegram plugin showed the same error as before.

ido1990, 2017-03-23:
Hmm,
I just uninstalled TouchUI but it didn't fix it.
I get this error on the log:
ERROR - Exception caught! 'NoneType' object has no attribute 'commandDict'

derpicknicker1, 2017-03-24:
That sounds strange. May you please enable debug option of plugin, try to reproduce the error and upload the octoprint.log to http://gist.github.com or somewhere else?

yorambi, 2017-09-23:
this may related to that in cmdSys @derpicknicker1 hard coded the system commands.
in windows install those do not exist by default and if added, they just being duplicated in the Telegram view. 

this should be removed from telegramCommands.py:

   `keys.append([["Restart OctoPrint","/sys_sys_Restart OctoPrint"]])`
   `keys.append([["Reboot System","/sys_sys_Reboot System"],["Shutdown System","/sys_sys_Shutdown System"]])`

this re-added:
`if empty: message += "\n\n"+self.gEmo('warning')+" No System Commands found..."`

as in v1.4.1 and same as in cmdCtrl

ir-fuel, 2017-11-11:
Same error here, same plugins installed. Going to try disabling some.

ir-fuel, 2017-11-11:
And yes, removing the room temperature plugin solved the issue

closed 2018-01-04 by derpicknicker1