jacopotediosi/OctoPrint-Telegram #75: Timelapse files support

funkycode, 2017-04-27:
I would like to request the feature to get timelapse files via chat commands.
I'll try to see if I can implement it by myself and PR, but as I'm not familiuar with OctoPi or Telegram APIs if I would be able to do it in short time period.
Thanks

derpicknicker1, 2017-06-06:
I have this on my list, but telegram does not support the video format of the timelapses. I figured out a command to properly convert them to use with telegram but i did not do further tests until now.

DaAwesomeP, 2017-06-08:
@derpicknicker1 For me, it would be fine if they were just available as downloads to view with another app (ex. VLC). At least please make it an option to download the original file; I'm not sure how well my OctoPrint machine will be able to re-encode the videos on the fly.

DeastinY, 2018-10-09:
Agreed. Just receiving the file would be amazing ❤️

danhae, 2019-07-31:
Hi, i would love to have this feature as well.
Maybe it is a solution to send it as a gif-file?
After the timelapse-video is finished, the plugin make it to a gif
`ffmpeg -ss 30 -t 3 -i input.mp4 -vf "fps=10,scale=320:-1:flags=lanczos,split[s0][s1];[s0]palettegen[p];[s1][p]paletteuse" -loop 0 output.gif` and send it via telegram.
Maybe it is a solution?

TheAndi, 2020-03-30:
Any news on that?
This feature would be awesome!

Thanks in advance

closed 2021-06-14 by giloser