jacopotediosi/OctoPrint-Telegram #76: Custom command to control ATX Power Supply?

poorredneck, 2017-05-03:
Hi, new to github so if this has already been addressed I apologize.

Is there a way to add custom /commands? Either executing a shell command or an OctoPrint system command?

I'm using the PSU Plugin that uses the GPIO write shell command.

I'd like to send /poweron or /poweroff and have it either 

A) issue a shell command like $ gpio -1 write 37 1 
B) Issue an existing octoprint command that basically does the same thing.

Thank you!

poorredneck, 2017-05-03:
Whoops! Looks like the /sys command does pring up my PSU controls!!! Would still be cool to have a /poweron command specific for that command though :)
Atleast I can turn the printer on through telegram though!

derpicknicker1, 2017-06-06:
Since there are many different ways for these commands depending on users printer setup, it wont work with a simple _/poweron_ command. I think the _/sys_ command is the best way for everyone.

closed 2017-06-06 by derpicknicker1

poorredneck, 2017-06-06:
You are absolutely correct, and I completely missed that there was a /sys
command. It works flawlessly with my custom system commands! Thank you.