jacopotediosi/OctoPrint-Telegram #78: Custom command for own notifications

Leprechaun91, 2017-05-07:
Hey,

is it possible to add own notifications?
For example when preheating is done?

derpicknicker1, 2017-06-06:
Not until now. I'm working on the possibility to add notifications for all octoprint events found [here](http://docs.octoprint.org/en/master/events/index.html#available-events). But that will take some time. 
There is no event for _preheat done_ in this list. So it is not planned/possible to support this.

filipgoc, 2018-01-28:
Less cryptic oprions: (Of course with parameters that will make them harmless.)

M300 (beep alert)
G4 (dwell)

modem7, 2018-05-15:
Definitely would be useful - looking to use the tplink plugin in conjunction with the auto connect plugin. So would be awesome to receive a message when /dev/ttyACM0 is connected/disconnected (which I'm not currently getting at the moment - I only get a message when the pi switches off/on/restarts)