jacopotediosi/OctoPrint-Telegram #80: ERROR - Exception caught! unsupported operand type(s) for *: 'float' and 'NoneType'

Hilleskog, 2017-06-07:
Hello! First of all I wish to thank you for your great work! Installed this yesterday and I think it will become very helpful tool for me in the future!

However I have one problem! Whenever I use the command "/print" and select the file I have uploaded to my OctoPrint server nothing happens! The only thing i see is a loading wheel on the selection I made! So I looked att the shell where I run OctoPrint and noticed a error-message. So I enabled logging for OctoPrint-Telegram and confirmed the error-message was shown there as well! Here is the ENTIRE log file but the error is on the last rows! Thank you for your dedication!

`2017-06-06 19:13:14,735 - octoprint.server - INFO - ******************************************************************************
2017-06-06 19:13:14,736 - octoprint.server - INFO - Starting OctoPrint 1.3.4 (master branch)
2017-06-06 19:13:14,737 - octoprint.server - INFO - ******************************************************************************
2017-06-06 19:13:14,988 - octoprint.plugin.core - INFO - Loading plugins from /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/hilleskog/.octoprint/plugins and installed plugin packages...
2017-06-06 19:13:16,906 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2017-06-06 19:13:16,910 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2017-06-06 19:13:16,983 - octoprint.plugin.core - INFO - Loading plugins from /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/hilleskog/.octoprint/plugins and installed plugin packages...
2017-06-06 19:13:17,334 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2017-06-06 19:13:17,336 - octoprint.printer.profile - ERROR - Profile _default does not exist, creating _default again and setting it as default
2017-06-06 19:13:17,459 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/hilleskog/.octoprint/uploads...
2017-06-06 19:13:17,461 - octoprint.filemanager.storage - INFO - ... file metadata for /home/hilleskog/.octoprint/uploads initialized successfully.
2017-06-06 19:13:20,718 - octoprint.util.pip - INFO - Using "/home/hilleskog/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2017-06-06 19:13:23,906 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2017-06-06 19:13:23,907 - octoprint.util.pip - INFO - pip installs to /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages, --user flag needed => no, virtual env => yes
2017-06-06 19:13:23,910 - octoprint.plugin.core - INFO - Initialized 7 plugin implementation(s)
2017-06-06 19:13:23,971 - octoprint.plugin.core - INFO - 7 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/virtual_printer
2017-06-06 19:13:23,972 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2017-06-06 19:13:23,981 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2017-06-06 19:13:23,984 - octoprint.server - INFO - Reset webasset folder /home/hilleskog/.octoprint/generated/webassets...
2017-06-06 19:13:23,985 - octoprint.server - INFO - Reset webasset folder /home/hilleskog/.octoprint/generated/.webassets-cache...
2017-06-06 19:13:24,603 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on hilleskog-lubuntu for SSDP
2017-06-06 19:13:24,608 - octoprint.server - INFO - Listening on http://0.0.0.0:5000
2017-06-06 19:13:24,932 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json
2017-06-06 19:13:24,974 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.29s
2017-06-06 19:13:25,824 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2017-06-06 19:13:25,984 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from http://plugins.octoprint.org/notices.json
2017-06-06 19:13:26,013 - octoprint.plugins.announcements - INFO - Loaded channel _releases from http://octoprint.org/feeds/releases.xml in 0.43s
2017-06-06 19:13:26,372 - octoprint.plugins.announcements - INFO - Loaded channel _blog from http://octoprint.org/feeds/octoblog.xml in 0.26s
2017-06-06 19:13:26,546 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from http://plugins.octoprint.org/feed.xml in 0.082s
2017-06-06 19:13:26,965 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from http://octoprint.org/feeds/octopi.xml in 0.25s
2017-06-06 19:13:55,389 - octoprint.server.util.flask.PreemptiveCache - INFO - Adding entry for / and {'query_string': 'l10n=en', 'path': u'/', '_count': 1, '_timestamp': 1496769235.389471, 'base_url': 'http://192.168.1.209:5000/'}
2017-06-06 19:14:07,310 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.117
2017-06-06 19:14:07,632 - octoprint.plugins.cura - INFO - Path to CuraEngine has not been configured yet or does not exist (currently set to None), Cura will not be selectable for slicing
2017-06-06 19:17:26,658 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2017-06-06 19:28:33,445 - octoprint.server - INFO - ******************************************************************************
2017-06-06 19:28:33,461 - octoprint.server - INFO - Starting OctoPrint 1.3.4 (master branch)
2017-06-06 19:28:33,461 - octoprint.server - INFO - ******************************************************************************
2017-06-06 19:28:33,866 - octoprint.plugin.core - INFO - Loading plugins from /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/hilleskog/.octoprint/plugins and installed plugin packages...
2017-06-06 19:28:36,895 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2017-06-06 19:28:36,899 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2017-06-06 19:28:37,073 - octoprint.plugin.core - INFO - Loading plugins from /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/hilleskog/.octoprint/plugins and installed plugin packages...
2017-06-06 19:28:37,434 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2017-06-06 19:28:37,493 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/hilleskog/.octoprint/uploads...
2017-06-06 19:28:37,514 - octoprint.filemanager.storage - INFO - ... file metadata for /home/hilleskog/.octoprint/uploads initialized successfully.
2017-06-06 19:28:37,570 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2017-06-06 19:28:41,552 - octoprint.util.pip - INFO - Using "/home/hilleskog/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2017-06-06 19:28:44,755 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2017-06-06 19:28:44,757 - octoprint.util.pip - INFO - pip installs to /home/hilleskog/OctoPrint/venv/lib/python2.7/site-packages, --user flag needed => no, virtual env => yes
2017-06-06 19:28:44,760 - octoprint.plugin.core - INFO - Initialized 7 plugin implementation(s)
[... cut ...]
2017-06-07 18:23:04,371 - octoprint.plugins.telegram.listener - INFO - Got a command: '/print' with parameter: '0bca257b|0|cbbfb4eba3ae57dfcfce0fa972118ed6' in chat 359674916
2017-06-07 18:23:04,756 - octoprint.plugins.telegram.listener - ERROR - Exception caught! unsupported operand type(s) for *: 'float' and 'NoneType'
2017-06-07 18:23:04,758 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 975386422...
2017-06-07 18:23:09,203 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAA1aAaaAaaa1aaaa_a1a1aAAAAAaaaaAaa
`

I would like to dig into it myself but I am very unfamiliar with this kind of coding!

derpicknicker1, 2017-06-07:
I can not reproduce this bug. May you have installed the _Cost Plugin_ ?

Hilleskog, 2017-06-07:
Yes I have! Is it the one causing problems?

derpicknicker1, 2017-06-07:
Is it configred? Do you have entered values for _Cost per hour_ and _Cost per meter_ ?

derpicknicker1, 2017-06-07:
I suspect there are no values configured.
When i save the _Cost Plugin_ settings with no values, i can reproduce the error.

Hilleskog, 2017-06-07:
I have configured it with settings and saved it! Will soon try uninstalling Cost Plugin and see if it helps!

derpicknicker1, 2017-06-07:
When you have entered values for Cost per hour and Cost per meter it should work fine

voodoobg, 2017-08-15:
I have the same bug. The thing is that i've filled all the values of the cost plugin and still Telegram wont initiate the print when I click on the file. I've disabled the cost plugin and it works fine.

guysoft, 2017-10-01:
Confirm here, the cost plugin is breaking it. Still relevant

chwinter82, 2017-10-01:
if you deactivate the cost plugin, everything is fine.

guysoft, 2017-10-01:
that's sorta what iI said :)

chwinter82, 2017-10-01:
sorry, just wanted to confirm.

closed 2019-08-07 by derpicknicker1