jacopotediosi/OctoPrint-Telegram #81: compatibility with OctoPrint PSU Control

leobel96, 2017-07-08:
Could you please add the possibility to switch on/off the printer adding the compatibility with OctoPrint PSU Control plugin (https://github.com/kantlivelong/OctoPrint-PSUControl/blob/master/README.md)? I think that adding the options in /sys should be perfect.

derpicknicker1, 2017-07-08:
The PSU Control plugin has no helper methods defined for use by other plugins.

leobel96, 2017-07-08:
So there is no possibility to switch on/off printer from telegram bot? What about the integration with other plugins that permit the relay control such as "Octoprint-enclosure" plugin?

derpicknicker1, 2017-07-08:
You may define own system commands (which e.g. call ascript). They will appear in /sys

leobel96, 2017-07-08:
Ok. Thank you

closed 2017-07-10 by derpicknicker1