jacopotediosi/OctoPrint-Telegram #82: Add support for viewing files on printer SD card?

daniel871, 2017-07-12:
Unless I'm missing a setting somewhere, this plugin cannot "see" the files located on the SD card of the printer the same way regular Octoprint access via web-browser can.

If I'm missing the setting, please tell me because it's bugging me that I can't /print (file) as the file I want to print doesn't show up when querying the telegram bot.

derpicknicker1, 2017-07-12:
SD card support is not build in at the moment...but on todo list

jacopotediosi, 2025-08-28:
The current implementation of `/files` already supports multiple [StorageInterface](https://docs.octoprint.org/en/master/modules/filemanager.html#octoprint.filemanager.storage.StorageInterface). In fact, if more than one is detected, the user is prompted to select one:  
https://github.com/jacopotediosi/OctoPrint-Telegram/blob/e4ec6fca51791d1aa0306c03387418a50c23a49f/octoprint_telegram/telegram_commands.py#L599-L613

Unfortunately, at the moment OctoPrint does not expose the SD card storage to plugins. I see that there is an open issue about it [here](https://github.com/OctoPrint/OctoPrint/issues/851), and that someone was working on its implementation years ago in [this PR](https://github.com/OctoPrint/OctoPrint/pull/889).