jacopotediosi/OctoPrint-Telegram #84: [Feature Request] Enclosure plugin integration

vitormhenrique, 2017-07-13:
Hello,

I'm the author of the octoprint-enclosure plugin. Some users have requested be to integrate my plugin with yours, so far I had no idea on how to implement such a thing, but Gina (one of the authors of octoprint) helped out and showed me the light at the end of the tunnel.

First, explaining what my plugin does: it is meat to be used on raspberry pi to control it's IO, using as input or outputs. You can control lights, fans, external heaters, add RGB led strips. Inputs (like buttons, or smoke sensors) can be used to control outputs, you can add external temperature sensors for measure and control the temperature of the enclosure, etc....

Everything documented here: https://github.com/vitormhenrique/OctoPrint-Enclosure

So few users asked me for the ability to have some of that functionality using the Telegram plugin. Instead of forking and merging and maintaining both of the plugins, Gina gave me a much better solution. Telegram can create a custom hook, to provide additional commands or extend status reports.

So you could create hooks asking for the pin setups, pin status, setting pin status etc, and I could implement those hooks on my plugin. That would make it possible for someone to basically use the octoprint-enclosure plugin using the telegram app.

You can see the original request here: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/48

Best regards

derpicknicker1, 2018-01-04:
I do think about hooks for a long time. But it will be a lot work to implement them correctly. I can't promise wether or when i will do this.
But it will be a great feature. Thanks for your inquiry.

kaju666, 2018-02-01:
Hello, any progress in that integration?

xinuxnt, 2019-10-10:
Hello, any progress?  I use both, your telegram plugin and enclosure, would be great to see both working together. I would help but my programming skills are flat. :-(

rlogiacco, 2019-10-11:
I guess this is a good time to raise those skills then...

As I don't have an enclosure, I have no need for such addition, thus.... 😄 
Maybe someone else is willing to pick this task, but my advice is to challenge yourself on the holy path of programming 😉

rbonghi, 2019-11-18:
Hi guys,

I follow up this feature. Should be really useful to manage remotely the integrations from Octoprint-Enclosure.

I tried to look the source code, but in a first look it's seems not simple to write this update. There is a short guide or example to help for this issue?

skandinaff, 2020-11-02:
Hello,

Just an idea, would it be possible to make it work through control commands? Maybe it would be easier than implement hooks in telegram plugin?

bluecraank, 2020-12-17:
Hello. this feature would be incredible. How it can be made? Maybe i can help.

TheBolzer, 2023-04-22:
Hey there,
I wanted to switch on my Lights(inside the Enclosure-plugin) via Telegram-plugin in, too.
And now I had found a solution, which works for me. 
My aim was to prepare my Printer for the next status-Picture and turn on the light.

My solution was in first place  the /gcode-Command inside the telegram-bot, to send the specific Enclosure-GCODE
like:
`/gcode_ENC O1 S1`  
this will switch on the first Output in the Enclosure-Plugin.

After some testing and searching i found this Post for automatic Pre-Snapshot activation and a little Timeout:
https://github.com/fabianonline/OctoPrint-Telegram/issues/358#issuecomment-1002769653

and so i also added a pre- and Post-routine for /status-Command:
Pre-Image-Action: as System-Command
`curl -X PATCH -H 'Content-Type: application/json' -H "X-Api-Key: xxxx-YOURAPI-KEY-xxxx" -i 'http://localhost/plugin/enclosure/outputs/1' --data '{ "status":true }' && sleep 2`

Post-Image-Action: as GCODE-Command:
`/gcode_ENC O1 S0`  

Why did i use System-Command for the Pre-Action?
--> With this i could use this "sleep"-function and it won't interrupt my print 

Hope this helps
markus

cesaregiannetti, 2025-07-17:
you can define system commands in `~/.octoprint/config.yaml` as documented [here](https://docs.octoprint.org/en/master/configuration/config_yaml.html#system) or use a [plugin](https://plugins.octoprint.org/plugins/systemcommandeditor/)
and there put your curl command or whatever you need to turn on the lights (I use OctoRelay API).
System commands are exposed to telegram via `/sys' command, so you can run them directly.