jacopotediosi/OctoPrint-Telegram #87: The current connection status is: Unknown

PifPof73, 2017-08-03:
I have made a new bot in telegram according to the documentation.
After type in of the API key, I get green information: "Token valid for @PifsTopeedobot." (name of my bot), but the connection status stays Unknown.

I can send messages to the bot and get the answer "Now I know you ...", but if I press reload or activate the reload button, I don't see any user in the chat list.

I have tried to do a ./pip install --upgrade https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip on my raspberry, which updated a few packages in the dependencies, and restarted the OctoPrint Server several times, but it seems that the telegram listener does not connect.

Where do I find log files? Can I test the listener without OctoPrint?

PifPof73, 2017-08-04:
I have activated the debug.log and found this entries related to telegram:

`2017-08-04 15:37:25,433 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2017-08-04 15:37:49,763 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAAAAaAaa1aaaAaA1aAaaAaAA1Aa1a1aAa
2017-08-04 15:37:55,788 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2017-08-04 15:38:26,056 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2017-08-04 15:38:56,397 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...`

The testing token looks strange - my token is different, but the Telegram Botname is loaded and shown in green.

PifPof73, 2017-08-04:
After resending some initial message I get the message I know you, and the log files says: there is a new user - but I don't see any in the web gui:

`2017-08-04 15:46:39,266 - octoprint.plugins.telegram - DEBUG - Sending a message: ℹNow I know you. Before you can do anything, go to OctoPrint Settings and edit some rights. with_image=False chatID= 44581
4867
2017-08-04 15:46:39,268 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True}
2017-08-04 15:46:39,269 - octoprint.plugins.telegram - DEBUG - Sending without image.. 445814867
2017-08-04 15:46:39,613 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
2017-08-04 15:46:39,615 - octoprint.plugins.telegram - DEBUG - Requesting Profile Photo for chat_id: 445814867
2017-08-04 15:46:39,895 - octoprint.plugins.telegram - DEBUG - NO PHOTOS 445814867. EXIT
2017-08-04 15:46:39,896 - octoprint.plugins.telegram.listener - DEBUG - Got new User
2017-08-04 15:46:39,896 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 349132514...
2017-08-04 15:47:10,194 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 349132514...
2017-08-04 15:47:40,488 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 349132514...
2017-08-04 15:48:10,812 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 349132514...`

"sending request ..." seems to be in an endless loop.

PifPof73, 2017-08-04:
Hi,
good news - I have found the reason: I had installed the plugin "Cyborg Theme (0.1.2)", which is a dark theme.
After deactivating the plugin everything worked well - the status is connected and after the initial ping my user was shown.

Sorry for opening the issue, but maybe it might be interesting for others, too.

Now I can say, that your work is great! Using a telegram bot seems to be more elegant than putting the complete raspberry through the firewall into the internet.
Greetings,
PifPof

closed 2017-08-04 by PifPof73

minusmagis, 2017-11-13:
Maybe it's a bit too late but I found out that it might be an incompatibility between plugins. I did not have the Cyborg theme plugin but I did have a bunch of other plugins active and when I deactivated them all It started working. I will reactivate them one by one to see which one might have been interfering.

petonic, 2018-03-18:
I think it was the indoor temperature plugin.  I deactivated and deleted that one, and it now works.  Good luck!