jacopotediosi/OctoPrint-Telegram #89: Telegram not responding when selecting file after "/file" command

chwinter82, 2017-08-19:
Hi,
I have two octopis running (on two different PIs).
Both were running fine and now I have trouble with just one:
when I enter "/files", the files are listed, but when I want to select a file (for printing e.g.), then nothing happens.
I already deactivated the telegram plugin and also reinstalled it, but with no success.

chwinter82, 2017-08-30:
Hi,
as I'm not receiving any answers yet, what would a way to solve the issue? reinstall the Pi completely new?

yorambi, 2017-09-29:
in OctoPrint setting on the Telegram plugin at bottom part, set "Enable logging of debug information for this plugin to octoprint.log"
then check the messages printed there to possibly find the issue...
the lines start with octoprint.plugins.telegram.

chwinter82, 2017-10-01:
Just found out via the log file and then googling the error line:
cost plugin is causing this issue. you have to setup the cost plugin correctly or deactivate the cost plugin. 
after that, telegram plugin works without any problem.

thanks @all for the help.

yorambi, 2017-10-01:
the telegram plugin currently (1.43) do not support a not configured cost plugin 
as the code in telegramCommands.py do not check for empty values in the two cost values

two functions fileDetails and fileOption have the below code with **cpM** , **cpH** and **curr** if empty will fail the **/file** command
```
cpH = self.main._settings.global_get_float(["plugins","cost","cost_per_hour"])
cpM = self.main._settings.global_get_float(["plugins","cost","cost_per_meter"])
curr = self.main._settings.global_get(["plugins","cost","currency"])
try:
	curr = curr.decode("utf-8")
except Exception, ex:
	pass
msg += "\n<b>Cost:</b> "+curr+"%.02f " % ((filaLen/1000) * cpM + (printTime/3600) * cpH)
```	
see issue #80

derpicknicker1, 2018-01-04:
This Issue will be fixed in next update. Will come the next days.

jlsjonas, 2019-04-09:
Hi, any update on this?

giloser, 2019-05-07:
Hi,

I think the problem is that the cost plugin change the variable name
we should change this
cpH = self.main._settings.global_get_float(["plugins","cost","cost_per_hour"])
cpM = self.main._settings.global_get_float(["plugins","cost","cost_per_meter"])
curr = self.main._settings.global_get(["plugins","cost","currency"])

into something like
cpH = self.main._settings.global_get_float(["plugins","cost","cost_per_time"])
cpM = self.main._settings.global_get_float(["plugins","cost","cost_per_length"])
curr = self.main._settings.global_get(["plugins","cost","currency"])

maybe add a try and except to avoid new trouble if the plugin change again.

PR #173 (2019-07-16): Update cost plugin vars
Updates cost plugin's variables, fixes #89 & likely #80
commit 10b8eef5 (2019-07-16): Update cost plugin vars

Updates cost plugin's variables, fixes #89 & likely #80
closed 2019-07-16 by rlogiacco