jacopotediosi/OctoPrint-Telegram #90: No Known Chats

PyroPony, 2017-08-24:
Hi,
I'm pretty new to this plugin on Octoprint but at the moment after I set the Telegram Bot Token, test it, and I send to my bot a message on Telegram. I have no Known Chats after I click to reload and if I set an auto refresh.
 Any help ?

maurocordioli, 2017-08-26:
me too, (OctoPrint: 1.3.4 , from octopi , plugin installed with plugin manager, if that helps)

Komaandy, 2017-08-27:
...same here... and i keep getting the replay in telegramm, that he "doesnt understand me"

PyroPony, 2017-08-27:
In telegramm, I have the answer "Now I know you. Before you can do anything, go to Ocotprint settings and edit somes rights." But I have nothing in the telegram plugin ...

criba, 2017-08-28:
Same here.
Is there a fix yet?

raphael-gatti-movile, 2017-08-30:
Hello Guys, same here.

I tried to check the log generated I even see that the user is been added to the local store, the picture is downloaded.

satanin, 2017-09-10:
Hi, same here, tried to uninstall the bot and repeat the process with same luck.

Steff1990, 2017-09-30:
Same here and still no good news!

yorambi, 2017-10-02:
in OctoPrint setting on the Telegram plugin at bottom part, set "Enable logging of debug information for this plugin to octoprint.log"
then check the messages printed there to possibly find the issue...
the lines start with octoprint.plugins.telegram.
will be useful if you share this log...

benkej05, 2017-10-13:
Here are my log. Fresh install, everything is up to date, plugin not working.

2017-10-13 22:55:47,522 - octoprint.plugins.telegram.listener - DEBUG - MESSAGE: {u'message': {u'date': 1507935347, u'text': u'X', u'from': {u'first_name': u'Janos', u'last_name': u'Benke', u'is_bot': False, u'id': 461252734, u'language_code': u'hu-HU'}, u'message_id': 7, u'chat': {u'first_name': u'Janos', u'last_name': u'Benke', u'type': u'private', u'id': 461252734}}, u'update_id': 159366449}
2017-10-13 22:55:47,525 - octoprint.plugins.telegram - DEBUG - Sending a message: â„ąNow I know you. Before you can do anything, go to OctoPrint Settings and edit some rights. with_image=False chatID= 461252734
2017-10-13 22:55:47,526 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True}
2017-10-13 22:55:47,527 - octoprint.plugins.telegram - DEBUG - Sending without image.. 461252734
2017-10-13 22:55:47,824 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
2017-10-13 22:55:47,827 - octoprint.plugins.telegram - DEBUG - Requesting Profile Photo for chat_id: 461252734
2017-10-13 22:55:48,078 - octoprint.plugins.telegram - DEBUG - NO PHOTOS 461252734. EXIT
2017-10-13 22:55:48,079 - octoprint.plugins.telegram.listener - DEBUG - Got new User
2017-10-13 22:55:48,079 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 159366450...
2017-10-13 22:55:57,205 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAA1AAaaaA1AAa1AAaAaaaAAAA11aAaaAa1
2017-10-13 22:56:18,347 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 159366450...
2017-10-13 22:56:30,372 - octoprint.plugins.telegram.listener - DEBUG - MESSAGE: {u'message': {u'date': 1507935390, u'text': u'G', u'from': {u'first_name': u'Janos', u'last_name': u'Benke', u'is_bot': False, u'id': 461252734, u'language_code': u'hu-HU'}, u'message_id': 9, u'chat': {u'first_name': u'Janos', u'last_name': u'Benke', u'type': u'private', u'id': 461252734}}, u'update_id': 159366450}
2017-10-13 22:56:30,375 - octoprint.plugins.telegram - DEBUG - Sending a message: â„ąNow I know you. Before you can do anything, go to OctoPrint Settings and edit some rights. with_image=False chatID= 461252734
2017-10-13 22:56:30,376 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True}
2017-10-13 22:56:30,377 - octoprint.plugins.telegram - DEBUG - Sending without image.. 461252734
2017-10-13 22:56:30,633 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
2017-10-13 22:56:30,637 - octoprint.plugins.telegram - DEBUG - Requesting Profile Photo for chat_id: 461252734
2017-10-13 22:56:30,883 - octoprint.plugins.telegram - DEBUG - NO PHOTOS 461252734. EXIT
2017-10-13 22:56:30,884 - octoprint.plugins.telegram.listener - DEBUG - Got new User
2017-10-13 22:56:30,884 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 159366451...
2017-10-13 22:56:31,412 - octoprint.plugins.telegram.listener - DEBUG - MESSAGE: {u'message': {u'date': 1507935391, u'text': u'V', u'from': {u'first_name': u'Janos', u'last_name': u'Benke', u'is_bot': False, u'id': 461252734, u'language_code': u'hu-HU'}, u'message_id': 11, u'chat': {u'first_name': u'Janos', u'last_name': u'Benke', u'type': u'private', u'id': 461252734}}, u'update_id': 159366451}
2017-10-13 22:56:31,415 - octoprint.plugins.telegram - DEBUG - Sending a message: â„ąNow I know you. Before you can do anything, go to OctoPrint Settings and edit some rights. with_image=False chatID= 461252734
2017-10-13 22:56:31,417 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True}
2017-10-13 22:56:31,417 - octoprint.plugins.telegram - DEBUG - Sending without image.. 461252734
2017-10-13 22:56:31,711 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
2017-10-13 22:56:31,714 - octoprint.plugins.telegram - DEBUG - Requesting Profile Photo for chat_id: 461252734
2017-10-13 22:56:31,950 - octoprint.plugins.telegram - DEBUG - NO PHOTOS 461252734. EXIT
2017-10-13 22:56:31,951 - octoprint.plugins.telegram.listener - DEBUG - Got new User
2017-10-13 22:56:31,952 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 159366452...
2017-10-13 22:56:32,308 - octoprint.plugins.telegram.listener - DEBUG - MESSAGE: {u'message': {u'date': 1507935392, u'text': u'G', u'from': {u'first_name': u'Janos', u'last_name': u'Benke', u'is_bot': False, u'id': 461252734, u'language_code': u'hu-HU'}, u'message_id': 13, u'chat': {u'first_name': u'Janos', u'last_name': u'Benke', u'type': u'private', u'id': 461252734}}, u'update_id': 159366452}
2017-10-13 22:56:32,311 - octoprint.plugins.telegram - DEBUG - Sending a message: â„ąNow I know you. Before you can do anything, go to OctoPrint Settings and edit some rights. with_image=False chatID= 461252734
2017-10-13 22:56:32,312 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True}
2017-10-13 22:56:32,313 - octoprint.plugins.telegram - DEBUG - Sending without image.. 461252734
2017-10-13 22:56:32,589 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
2017-10-13 22:56:32,592 - octoprint.plugins.telegram - DEBUG - Requesting Profile Photo for chat_id: 461252734
2017-10-13 22:56:32,823 - octoprint.plugins.telegram - DEBUG - NO PHOTOS 461252734. EXIT
2017-10-13 22:56:32,824 - octoprint.plugins.telegram.listener - DEBUG - Got new User
[... cut ...]
2017-10-13 22:59:46,740 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 159366459...
2017-10-13 23:00:17,022 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 159366459...
2017-10-13 23:00:36,338 - backoff - ERROR - Backing off capture_mjpeg(...) for 374.2s (URLError: <urlopen error [Errno 111] Connection refused>)
2017-10-13 23:00:47,338 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 159366459...

yorambi, 2017-10-15:
you have to follow the [User setup ](https://github.com/fabianonline/OctoPrint-Telegram#user-setup) instructions... when your chat line show, you have to click the icons under the Command, Notify, Action column to set the permissions for [Users](https://github.com/fabianonline/OctoPrint-Telegram#users)...

bzng, 2017-10-24:
Still same problem. Bot is fine. Octoprint is fine. Token seems fine. OctoLog seems fine. But there arent and 'Known Chats'.

2017-10-24 17:32:51,390 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 485700249...
2017-10-24 17:33:22,013 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 485700249...
2017-10-24 17:33:52,594 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 485700249...
2017-10-24 17:34:12,223 - octoprint.plugins.telegram.listener - DEBUG - MESSAGE: {u'message': {u'date': xxxxx, u'text': u'333', u'from': {u'first_name': u'Simon', u'is_bot': False, u'id': xxxxx, u'language_code': u'de'}, u'message_id': 47, u'chat': {u'first_name': u'Simon', u'type': u'private', u'id': xxxxx}}, u'update_id': xxxxx}
2017-10-24 17:34:12,226 - octoprint.plugins.telegram - DEBUG - Sending a message: ℹNow I know you. Before you can do anything, go to OctoPrint Settings and edit some rights. with_image=False chatID= xxxxx
2017-10-24 17:34:12,228 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True}
2017-10-24 17:34:12,229 - octoprint.plugins.telegram - DEBUG - Sending without image.. 10915893
2017-10-24 17:34:12,926 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
2017-10-24 17:34:12,928 - octoprint.plugins.telegram - DEBUG - Requesting Profile Photo for chat_id: 10915893
2017-10-24 17:34:13,475 - octoprint.plugins.telegram - DEBUG - Requesting file with id AgADBAADqqcxGzWQpgABrwKv1xA4qOKsiIQrAATJ6LYIo0J9qFlwAAIC.
2017-10-24 17:34:14,064 - octoprint.plugins.telegram - DEBUG - Downloading file: https://api.telegram.org/file/botxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx/photos/file_0.jpg
2017-10-24 17:34:14,636 - octoprint.plugins.telegram - DEBUG - Saved Photo 10915893
2017-10-24 17:34:14,638 - octoprint.plugins.telegram.listener - DEBUG - Got new User
2017-10-24 17:34:14,639 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 485700250...

runkkusankari, 2017-11-03:
Hi,

I have same problem. I get message from the bot "Now i know you. Before you can do.." Token says valid for my bot but i cant get any known chats. Current connection status says "Unknown".

yorambi, 2017-11-08:
Did you followed [step 5](https://github.com/fabianonline/OctoPrint-Telegram#optional) ?  looks that this is probably **not**  _"**OPTIONAL**"_...

> 5. Tell the botfather which commands are available. This enables Telegram to auto-complete commands to your bot. Send /setcommands to @botfather, select the bot and then send the lines in the box below (one message with multiple lines).
> 
> /setcommands
> abort - Aborts the currently running print. A confirmation is required.
> shutup - Disables automatic notifications till the next print ends.
> dontshutup - The opposite of /shutup - Makes the bot talk again.
> status - Sends the current status including a current photo.
> settings - Displays the current notification settings and allows you to change them.
> files - Lists all the files available for printing.
> print - Lets you start a print. A confirmation is required.
> togglepause - Pause/Resume current Print.
> con - Connect/disconnect printer.
> upload - You can just send me a gcode file to save it to my library.
> sys - Execute Octoprint System Comamnds.
> ctrl - Use self defined controls from Octoprint.
> tune - Set feed- and flowrate. Control temperatures.
> user - get user info.
> help - show list of commands.

clihlt, 2017-11-09:
Same problem. There are three users. I've seen them before. I gave them the rights and permissions. But now there are no known users. Just an empty list. But those three users are active and working. I can not change or delete them. I can not add new users. They receive a greeting from the bot, but they do not appear in the list.

derpicknicker1, 2018-01-04:
This is most likely a problem woth an other plugin: Octoprint Anywhere. Many users reported this issue and deactivateing/uninstalling octoprint anywhere helped.
It seems that the new octoprint version (1.3.6) fixed this issue.

Poikilos, 2023-01-03:
Seeing as so many people have this problem, maybe the steps should be easier than:
1. There is no error, but enter an API key make the innocuous pencil appear
2. Click an innocuous pencil, reload, wait for more innocuous symbols to appear. It may say to check the log, which is misleading if there is no API key at all.
3. click each innocuous symbol and enable things

If any step is missed, plugin won't work.

I suggest something bigger and bolder such as in these mockups:
1.
[image]
(instead of saying:
> in OctoPrint setting on the Telegram plugin at bottom part, set "Enable logging of debug information for this plugin to octoprint.log" then check the messages printed there to possibly find the issue... the lines start with octoprint.plugins.telegram. will be useful if you share this log...
)
- @yorambi 

2.
[image]
3.
[image]

All these people (and me, programmer for 28 years) can't be assumed to be just not being smart enough.

> In telegramm, I have the answer "Now I know you. Before you can do anything, go to Ocotprint settings and edit somes rights." But I have nothing in the telegram plugin ...

- @oreldx

> Same here. Is there a fix yet?

- @ghost

> Still same problem. Bot is fine. Octoprint is fine. Token seems fine. OctoLog seems fine. But there arent any 'Known Chats'.
> 

- @bzng 

> 

> Hi,
> 
> I have same problem. I get message from the bot "Now i know you. Before you can do.." Token says valid for my bot but i cant get any known chats. Current connection status says "Unknown".

- @runkkusankari 

I think there will be far fewer log dumps in the issue board if the UI explains what is actually wrong by detecting these 3 situations above (in my UI mockups) instead of saying to check the log.

If you're concerned there are various people adding things to this issue that may not fit the issue, let me suggest that adding these messages will also make clear what state the plugin is in. I can't tell if my issue is the same or not, but it was the same as all of these people I quoted and it was solved by following the steps that I've hopefully made more clear. Making the documentation clear hasn't been enough, if the steps themselves are unusual and there is no clear indication of state within the UI.

jacopotediosi, 2025-07-10:
I agree with @Poikilos; most likely all the above issues stem from user errors. Actually, the workflow and UI/UX for configuring this plugin seem too complicated. There is definitely room for improvement here.