jacopotediosi/OctoPrint-Telegram #92: Command failed with exception: 'NoneType' object is not iterable!

yorambi, 2017-09-23:
in telegramCommands.py -> cmdSys @derpicknicker1 hard coded the system commands.
on windows install those do not exist by default and if added, they just being duplicated in the Telegram view. 

this should be removed from telegramCommands.py:

   `keys.append([["Restart OctoPrint","/sys_sys_Restart OctoPrint"]])`
   `keys.append([["Reboot System","/sys_sys_Reboot System"],["Shutdown System","/sys_sys_Shutdown System"]])`

this re-added:
`if empty: message += "\n\n"+self.gEmo('warning')+" No System Commands found..."`

as in v1.4.1 and same as in cmdCtrl 

error in log:
octoprint.plugins.telegram.TCMD - WARNING - Command failed: 'NoneType' object is not iterable

CoolSilver, 2017-11-10:
I had to revoke my token, it made a new one and this resolved the issue. 

Also had this issue with the Cyborg theme which created a similar error which I had to correct first.
Token status is: Error: 'NoneType' object has no attribute 'username'

https://github.com/fabianonline/OctoPrint-Telegram/issues/87#issuecomment-320269641

CoolSilver, 2017-11-11:
Nope still broke, nothing sends and reboot goes back to way it was before

derpicknicker1, 2018-01-05:
@yorambi May you do a pull request with your bugfix?

yorambi, 2018-01-11:
Done...

Lepterois, 2018-01-11:
I have the same issue since yesterday. I tried even to delete the bot and make a new bot, that the same issue

closed 2025-07-10 by jacopotediosi