jacopotediosi/OctoPrint-Telegram #93: Error of notifications

garremmash, 2017-10-20:
Here is such problems

ERROR: I was not able to format the Notification for 'PrinterStart' properly. Please open your OctoPrint settings for Telegram Notifications and check message settings for 'PrinterStart'.

ERROR: I was not able to format the Notification for 'PrinterShutdown' properly. Please open your OctoPrint settings for Telegram Notifications and check message settings for 'PrinterShutdown'.

But on the settings everything is ok.

derpicknicker1, 2018-01-04:
Please post the message text from the settings.

Evg33, 2019-05-14:
`{emo:octo} {emo:shutdown} Shutting down. Goodbye.`

`{emo:rocket} Hello. I'm online and ready to receive your commands.`

CapXilinx, 2019-05-15:
I have the same issue. It seems that this problem is related to the used emotes. When I deactivate the emotes the message are Sent correctly.

thesourcerer, 2019-05-15:
Got the same problem with "PrinterStart" after upgrading OctoPrint today to latest 1.3.11

miblafi, 2019-05-16:
Had the same issue, I disabled the Emojis and the message was successfully sent, as CapXilinx did

Evg33, 2019-05-16:
After Octoprint upgrade to 1.3.11 telegram plugin is unable to format emo:rocket and other emos #174

Budavaril, 2019-06-19:
Using no space or at least 2 spaces between rocket emo and the text. With 1 spaces will give error.

kverbeeck, 2019-07-02:
> Using no space or at least 2 spaces between rocket emo and the text. With 1 spaces will give error.

Thanks man, great workaround.

rlogiacco, 2019-07-17:
This has been fixed in the current master branch and will be delivered with the next release.

closed 2019-07-17 by rlogiacco