jacopotediosi/OctoPrint-Telegram #95: Doesn't work

ir-fuel, 2017-11-11:
I created the bot, added the commands, connected the chat, enabled everything in OctoPrint settings, but I never get any status updates.

Also, when sending `/status` to my bot it simply does not reply.

This is what's in the logs
```
2017-11-11 21:24:29,806 - octoprint.plugins.telegram.listener - DEBUG - MESSAGE: {u'message': {u'from': {u'first_name': u'Joris', u'last_name': u'Mans', u'is_bot': False, u'id': 480750888, u'language_code': u'en-BE'}, u'text': u'/status', u'entities': [{u'length': 7, u'type': u'bot_command', u'offset': 0}], u'chat': {u'first_name': u'Joris', u'last_name': u'Mans', u'type': u'private', u'id': 480750888}, u'date': 1510435469, u'message_id': 9}, u'update_id': 445843029}
2017-11-11 21:24:29,808 - octoprint.plugins.telegram.listener - INFO - Got a command: '/status' with parameter: '' in chat 480750888
2017-11-11 21:24:29,809 - octoprint.plugins.telegram.listener - ERROR - Exception caught! 'accept_commands'
2017-11-11 21:24:29,810 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 445843030...
```

Seems there is an error

gappleto97, 2017-11-14:
Having similar issues here as well. It doesn't even allow for `/start` messages anymore. Just responds either that I don't have permission, or that it doesn't understand.

derpicknicker1, 2018-01-04:
@ir-fuel maybe its an issue with your config file? Seems like the setting 'accept_commands' for your user is missing.

ir-fuel, 2018-01-04:
I don't know what config file you are talking about?
I uninstalled it because it doesn't work, and has compatibility issues with quite a few plugins.

derpicknicker1, 2018-01-04:
the config.yaml from octoprint

derpicknicker1, 2018-01-04:
On which plugins you observrd the incompability and what where the issues?

ir-fuel, 2018-01-04:
I never touched that file.

ir-fuel, 2018-01-04:
I don't remember everything (It's been a couple of months) but one was the room temperature plugin.

derpicknicker1, 2018-01-05:
I will check that

ir-fuel, 2018-01-05:
Tbh it's a bit of pity this doesn't work. It seems to be the only decent plugin to send status updates + take pictures. But the lack of support when I tried it (you can see the date of my post here + the date of your first response) made me abandon it because I simply could not get it to work. 
No worries, you're not obliged to anything of course, this is a free project so I have no right to complain. Just wanted to pass the message.

ir-fuel, 2018-01-06:
If you have time to debug some issues, I will be happy to try installing and running it again. Just let me know!
Thanks.

rlogiacco, 2019-07-17:
Is this issue still affecting the plugin? Closing it, but comment back and we will re-open it.

closed 2019-07-17 by rlogiacco