jacopotediosi/OctoPrint-Telegram #96: [Feature request] Make PIL/Pillow optional

Depaulicious, 2017-11-12:
I'm trying to run OctoPrint on PyPy on ARM and it's being a pain because this module, which I don't want to give up, depends on Pillow and it won't run on PyPy on ARM (I opened an issue https://github.com/python-pillow/Pillow/issues/2850).

I gave a look at the code and it seems to me that Pillow can be made optional. It's only used to flip the pictures from the webcam and to scale down profile pictures.

I think these features can be disabled if Pillow is not installed (original propic is used, no h/vflip feature)

What do you think?

derpicknicker1, 2018-01-04:
Thats an option to think about. But when i remove pillow from the plugin requirements, it will not automatically insatll pillow for users which can use it.

Depaulicious, 2018-01-08:
Users can simply install it manually. But I see some users might not know how to do it. Maybe we can just make an extra "fake" plugin that depends on OctoPrint-Telegram and Pillow so it gets pulled in.

Depaulicious, 2018-02-25:
Other alternate solutions for this issue that came to my mind in the last few days:

- Pillow can be made optional in the code but `setup.py` can still depend on it. Users that don't want to use it will have to handle the unresolved dependency with `pip` on their own.
- The plugin may detect the `pillow` ImportError and not use it.

I will try to implement one of these and send you a PR as I would like to use it in my setup, feel free not to merge it if you don't like the workarounds (I would not blame you for that, this is definitely not an OctoPrint-Telegram issue).

jacopotediosi, 2025-07-10:
Related: #22

jacopotediosi, 2025-07-20:
The ticket https://github.com/python-pillow/Pillow/issues/2850 was closed because the OP no longer has the device on which they were unable to install Pillow. Also, Pillow maintainers were never able to reproduce the bug, so it seems it was specific to the OP's environment.

I'm closing this issue but keeping #22 open to evaluate whether Pillow should be made optional or replaced with alternatives.

closed 2025-07-20 by jacopotediosi