jacopotediosi/OctoPrint-Telegram #100: Dual extruder support in status messages

mcamou, 2017-12-29:
I just installed a second extruder. The status messages only show the temperature of the first extruder:

```
Magrathea OctoPrint:
Printing at Z=10.95.
Bed 39.98/40.0, Extruder 44.0/0.0.
00:46:16, 66% done, 00:23:17 remaining.
```

This is while printing just with E2 (Tool 1), so E1 (Tool 0) is turned off.

derpicknicker1, 2018-01-03:
Change the message in the settings (ZChange, also for StatusPrinting).

The original message looks like this:
```
Printing at Z={z}.
Bed {bed_temp}/{bed_target}, Extruder {e1_temp}/{e1_target}.
{time_done}, {percent}% done, {time_left} remaining.
```
For the second extruder use `{e2_temp}` and `{e2_target}`. Yor message will look like the following if you want to get the status of both extruders:

```
Printing at Z={z}.
Bed {bed_temp}/{bed_target}
Extruder1 {e1_temp}/{e1_target}
Extruder2 {e2_temp}/{e2_target}
{time_done}, {percent}% done, {time_left} remaining.
```

derpicknicker1, 2018-01-03:
See also: https://github.com/fabianonline/OctoPrint-Telegram#notifications

closed 2018-01-03 by derpicknicker1