jacopotediosi/OctoPrint-Telegram #104: [Feature Request] Allow list of z-heights instead of intervals

brentmc79, 2018-01-23:
**Use Case**

- I have a print with filament changes at 10mm, 14mm, and 22mm
- I would like to be notified before the print gets to those z-heights
- Instead of set intervals, allow a list of z-heights for when to send messages

filipgoc, 2018-01-28:
Interesting idea. 
It makes me think of a more general solution to this... Why not have some sort of event/command to always send a status notification? I'll elaborate in another post.

filipgoc, 2018-01-28:
@brentmc79 you will be able to accomplish this with the generic feature "Generic GCODE command notification". It's in the 2.0 milestones.

milestones: https://github.com/fabianonline/OctoPrint-Telegram/milestone/4