jacopotediosi/OctoPrint-Telegram #105: not receive the message "I know You"

Lepterois, 2018-01-24:
Hi!
When I was using Octoprint-Telegram with the old version of Octoprint, everything was working fine.
I've done a new install with the last version of Octoprint and reinstall Octoprint-Telegram.
I was not able to receive the message "I know you". I tried to delete the bot and make a new one, same result. I see the bot in the list of chat, but I'm not able to edit the action with the pencil. When I clic on refresh it start to searching and no stop.
When I test the token, the answer is "token valid for @...bot".
I don't understand. Do I miss a dependence or something like this?
Help me please, this functionality was so good!

Thanks in advance

Lepterois, 2018-01-24:
Hi all!
Tonight I tried something... do a new installation on another SD card and just installing telegram. It was working perfectly.
And because normally I use my PI3 with a ssd drive, I tried to do the same but on the SSD drive. And ... It was not working! 
I suppose my problem come from the fact I use the PI from a SSD drive?
I will try to put my old installation from my SSD to the SD card to see if it's working or not...
to be followed....

Lepterois, 2018-01-24:
No success.... the same problem after recovery my SSD drive on the SD card.
I think the only solution I have now, it's to reinstall everything on the sd card :-(

digitalfrost, 2019-07-31:
Stale Ticket. This can probably be closed.

jacopotediosi, 2025-07-10:
No logs -> ticket closed

closed 2025-07-10 by jacopotediosi