jacopotediosi/OctoPrint-Telegram #109: Plugin error with custom system commands

mcamou, 2018-01-26:
I have a couple of custom system commands that work correctly on the web interface but not from the Telegram plugin:

```
system:
  actions:
  - action: printeron
    command: /home/pi/bin/printerOn
    confirm: false
    name: Printer on
  - action: printeroff
    command: /home/pi/bin/printerOff
    confirm: Are you sure you want to power off the printer?
    name: Printer off
```

This is the `printerOn` script (`printerOff` is similar):

```
#!/bin/sh
curl --header ''Content-Type: text/plain'' --data ON http://openhab.local:8080/rest/items/Magrathea_OnOff
sleep 10
$HOME/oprint/bin/octoprint client command api/connection connect
```

This is the error I get in `octoprint.log` with these commands (built-in commands like "Restart Octoprint" work fine):

```
2018-01-26 11:04:46,881 - octoprint.plugins.telegram.listener - INFO - Got a command: '/sys' with parameter: '9ed49207baf9d20a975a92c3f4b9005e' in chat 370432613
2018-01-26 11:04:47,208 - octoprint.plugins.telegram.listener - ERROR - Exception caught! coercing to Unicode: need string or buffer, bool found  
```

Bart1909, 2018-04-13:
I'm getting the same error :-(
`2018-04-13 16:12:06,391 - octoprint.plugins.telegram.listener - INFO - Got a command: '/sys' with parameter: '979e3897441fc29a2cc5635c3c36491e' in chat 55650497
2018-04-13 16:12:06,397 - octoprint.plugins.telegram.listener - ERROR - Exception caught! coercing to Unicode: need string or buffer, bool found`

My config.yaml looks like:
`system:
  actions:
  - action: lightson
    command: ssh -oStrictHostKeyChecking=no pi@pi2 /home/pi/schalter.sh 1
      ON
    confirm: false
    name: Lights On
  - action: lightsoff
    command: ssh -oStrictHostKeyChecking=no pi@pi2 /home/pi/schalter.sh 1
      OFF
    confirm: false
    name: Lights Off
`

numanair, 2018-10-16:
How can I use this fix on my own setup?

closed 2019-08-07 by derpicknicker1